realsenseai/librealsense #3184: Change Compiler for Cross Compiling librealsense + Change Installation Directory

abdelrahmanhussein, 2019-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435 | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |  Petalinux  | 
| Kernel Version (Linux Only)    | 4.14.0-xilinx-v2018.2 |                                         | 
| Platform                                 | Xilinx Ultra96 FPGA  |
| SDK Version                            | 2.17.0                         | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue Description

Hello,

First, I am using Ubuntu 16.04 with Kernel 4.15.0-43-generic

I am working on cross-compiling librealsense using these compilers aarch64-linux-gnu-gcc and aarch64-linux-gnu-g++ to install it on petalinux (Xilinx Embedded Linux Solution) root file system, and run it on the system with the specs shown in the required info section above. So, I need to configure the whole (cmake -> make -> install) flow with the following:
1) different compilers (ones stated above)
2) different installation location.

I have done some search and found out a similar case that use this cmake command `cmake ../ -DCMAKE_TOOLCHAIN_FILE=/home/user/hacking/aarch64.cmake` to configure it for the cross-compilation, where aarch64.cmake file is the following:

```
# this one is important
SET(CMAKE_SYSTEM_NAME Linux)
#this one not so much
SET(CMAKE_SYSTEM_VERSION 1)

# specify the cross compiler
SET(CMAKE_C_COMPILER   /usr/bin/aarch64-linux-gnu-gcc)
SET(CMAKE_CXX_COMPILER /usr/bin/aarch64-linux-gnu-g++)
SET(CMAKE_PREFIX_PATH /usr/lib/x86_64-linux-gnu/cmake/glfw3)

# where is the target environment
SET(CMAKE_FIND_ROOT_PATH /usr/aarch64-linux-gnu)

# search for programs in the build host directories
SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
# for libraries and headers in the target directories
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

#Find OPENGL
find_package(OpenGL REQUIRED)
set(DEPENDENCIES realsense2 glfw ${OPENGL_LIBRARIES})

# end of the file

```

After running this, I ended up with so many cmake environment variables that were undefined:

```
CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_PREFIXES
CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_SUFFIXES
CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_PREFIXES
CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_SUFFIXES
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindOpenGL.cmake:172 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /home/user/hacking/aarch64.cmake:21 (find_package)
  build/CMakeFiles/3.5.1/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:3 (project)

```

So:
1) How can I change the compiler in the CMake configuation file?
2) How can I change the installation directory to one of my choice?

RealSenseCustomerSupport, 2019-02-08:
----------------------------------------------

Hello @abdelrahmanhussein,

Regarding both of your issues:

-  https://github.com/IntelRealSense/librealsense/issues/3040
-  https://github.com/IntelRealSense/librealsense/issues/3184

We have not qualified Xilinx Ultra96 FPGA for LibRealSense / RealSense Cameras.

However, this is an ARM device, perhaps the instructions that are used for installing on a Raspberry PI3 will be helpful:  https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

Note that instructions to activate the OpenGL driver will most likely not apply to your device, you will need to research how to display video.

I suggest to try compiling on Xilinx Ultra96 FPGA device from scratch and not changing the compilers.

Please post your results compiling within your PetaLinux environment.

Thank you!

abdelrahmanhussein, 2019-02-08:
@RealSenseCustomerSupport 

Hello,

Thank you so much for your response.

Is not it possible to change the compilers as I asked in my previous post? I think CMake can be configured to include the aarch-64 compiler in its toolchain and use it to compile the library instead of the regular flow.

Thanks a lot.

RealSenseCustomerSupport, 2019-02-08:
----------------------------------------------

Hello again @abdelrahmanhussein,

Sorry the info was not helpful, here is some information on creating a cross-compilation environment with aarch64:

https://preshing.com/20141119/how-to-build-a-gcc-cross-compiler/

Please post your results after you setup environment and run compilation.  If successful this will be helpful to others attempting this same process.

Thank you!

abdelrahmanhussein, 2019-02-08:
Sure, I will give it a try and in case I managed to do it, I will surely share the details.

I will leave the inquiry open in case I need to get back to you with further inquiries.

Thanks a lot for your amazing support.

RealSenseCustomerSupport, 2019-02-23:
----------------------------------------------

Hello @abdelrahmanhussein,

Were you able to get a cross compilation environment to work?

If no response we will close issue.

Thank you!

abdelrahmanhussein, 2019-02-23:
Hello,

I am still working on it.

Will update the status within the next few hours.

closed 2019-02-24 by abdelrahmanhussein

thangbk2231, 2019-12-04:
Hi @abdelrahmanhussein 
Did your success to solve the problem?
Can you share how to do it?

Thank you!

abdelrahmanhussein, 2019-12-20:
Hey @thangbk2231 

Well, I spent a lot of time working on it. I can tell you about my work until the point I decided to stop since that was a very long time ago.

So, the problem I faced was how to cross-compile the entire driver and the SDK to run on the ARM Core on Xilinx UltraScale FPGA. This will take you to use the ARM toolchain installed with Xilinx SDK, by stating which compiler to use in the CMAKE GUI (using the commands was not really helpful as you will need to modify a lot of parameters for the CMAKE).

Take into consideration that you will need to also to cross-compile the entire dependencies.

Best case scenario is to have a Linux distribution running on your targeted device which has CMAKE package installed on it (in my case, I wasn't able to have a petalinux [xilinx linux solution] that has CMAKE installed).

Something was kinda helpful to me was to refer to a website (lost its URL unfortunately) where you may find the packages already cross-compiled, yet, some of them I failed to run them correctly on the targeted device.

abdelrahmanhussein, 2019-12-20:
@thangbk2231 You refer to this as well: https://github.com/IntelRealSense/librealsense/issues/2197