realsenseai/librealsense #3191: D400 family firmware v5.11.1.0 not supported

sogartar, 2019-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.1.0 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.10.0                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.0                          | 
| Language                            |  C                           | 

### Issue Description
v2.18.0 does not support the latest firmware v5.11.1.0 for D400 Family. It gives
```
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed. Error type: Invalid Command (-1).
```

You probably know that already, but I am submitting it anyway.

dorodnic, 2019-02-03:
Yes, we will release 2.18.1 shortly that will fix backward-compatibility with 5.11.1 and also we are waiting for the new firmware to be released that will be backward-compatible with 2.18.0 as well.

chetansarmak, 2019-02-05:
Hi dorodnic 
I am getting a error saying 

terminate called after throwing an instance of 'rs2::invalid_value_error'
what():  object doesn't support option #15
Aborted (core dumped)

In this #3185 when using the options of RS2_OPTION_MOTION_RANGE

Is it anything related to the firmware versions, to solve this error?

dorodnic, 2019-02-10:
Resolved in [v2.18.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1)

closed 2019-02-10 by dorodnic