realsenseai/librealsense #3193: aptrepo is not accessible on Ubuntu 18

cstathis, 2019-02-01:
Following up on this issue which was closed with no actual resolution.
https://github.com/IntelRealSense/librealsense/issues/2884

apt cannot pull from the realsense repository on Ubuntu 18. It looks like the repo is misconfigured.
This prevents me from installing the DFU tool.

Err:11 http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo xenial InRelease                       
  403  Forbidden [IP: 52.218.20.97 80]                                             
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.20.97 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo xenial InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

RealSenseCustomerSupport, 2019-02-07:
----------------------------------------------

Hello @cstathis,

     We just checked a clean install of Ubuntu 18.04 with these instructions: (Copy and paste these instructions into your terminal, using a non-corporate, commercial network,)

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list

sudo apt-key adv--keyserverkeys.gnupg.net --recv-key 6F3EFCDE

sudo apt-get update

sudo apt-get install intel-realsense-dfu*

Sources:
-  https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html <-- Linux DFU Instructions

-  https://downloadcenter.intel.com/download/28377/Latest-Firmware-for-Intel-RealSense-D400-Product-Family <-- Latest RealSense Camera firmware downloads.

     We were able to download and install Linux DFU on Ubuntu 18.  Issues you are seeing may have to do with your network.  Are you on a corporate network behind a firewall?  It is recommended to not use corporate firewalls unless you have the proper proxies enabled.

There is also a docker image available that contains the Linux DFU, built and tested on Ubuntu 18:

-  Install Docker using these instructions:  https://www.digitalocean.com/community/tutorials/how-to-install-and-use-docker-on-ubuntu-18-04

-  Pull docker container with preinstalled DFU:

sudo docker pull speechtesting11/ubu18_dfu1_lrs

-  Commands to run docker container:

xhost local:root [ENTER]

sudo docker run -it --volume=/tmp/.X11-unix:/tmp/.X11-unix --device=/dev/dri:/dev/dri --privileged -v /dev:/dev --env="DISPLAY" speechtesting11/ubu18_dfu1_lrs [ENTER]

-  Print Intel RealSense DFU version, type:

Intel-realsense-dfu -v

NOTE:  This docker container was created on an Intel NUC7i3BNH, so if you have errors running given instructions, most likely it has to do with hardware limitations.

Please give the above a try, making sure your network settings are not limited in any way.

Thank you!

cstathis, 2019-02-08:
In my earlier post I showed the 403 Forbidden errors I get during apt-get update. This is a response from your apt server, which leads me to believe the issue is on your side.

I confirmed with my IT department that no traffic was blocked between my machine to these endpoints by our firewall. If I browse to your AWS server in a web browser, I get the same kind of 403 server errors. 

It looks as if my IP address range is blacklisted by your network?

Note that I have the GPG key:

Executing: /tmp/apt-key-gpghome.SFo7prRvQV/gpg.1.sh --keyserver hkp://keys.gnupg.net:80 --recv-key 6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

RealSenseCustomerSupport, 2019-02-08:
----------------------------------------------

Hello @cstathis,

You mention that you have an IT department which means you are still in a corporate network environment.

Would it be possible to try a completely different network?  Maybe try enabling a mobile WIFI hot-spot with your phone, connect to that network and give the install a try.

Please update this ticket with your results.

Thank you

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

No response in 14 days.

Close ticket.

closed 2019-02-25 by dorodnic

spd-intermodalics, 2021-06-07:
Is this happening again?