realsenseai/librealsense #3197: Error Building for Jetson Xavier..

kooscode, 2019-02-01:
Camera Model = D415 
Operating System & Version = Ubuntu 18.04  
Kernel Version: 4.9.108-tegra
Platform = Jetson Xavier
SDK Version = Latest Git Pull on Master

### Issue Description
**Failed to make RealSense SDK**

usr/bin/ld: CMakeFiles/libtm_util.dir/libtm_util.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.17'
//lib/aarch64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
third-party/libtm/tools/libtm_util/CMakeFiles/libtm_util.dir/build.make:95: recipe for target 'third-party/libtm/tools/libtm_util/libtm_util' failed
make[2]: *** [third-party/libtm/tools/libtm_util/libtm_util] Error 1
CMakeFiles/Makefile2:1763: recipe for target 'third-party/libtm/tools/libtm_util/CMakeFiles/libtm_util.dir/all' failed
make[1]: *** [third-party/libtm/tools/libtm_util/CMakeFiles/libtm_util.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 59%] Built target realsense2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

kooscode, 2019-02-02:
As a workaround, I edited the _"CMake/unix_config.cmake"_ file and changed added pthread to C and CXX make flags.

That worked and I was able to build librealsense with CUDA and OpenMP support.

```
set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}  -pthread -mstrict-align -ftree-vectorize")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -mstrict-align -ftree-vectorize")  
```

gliese581gg, 2019-02-07:
Got the same error. Same workaround works. Thanks!!

closed 2019-02-26 by RealSenseCustomerSupport