realsenseai/librealsense #3198: D453i setup issues

smudgy67, 2019-02-01:
| | |
|---|---|
|**librealsense**|2.18.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|841512071497|
|**Firmware Version**|05.11.01.00
255.255.255.255|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.0|

Please provide a description of the problem:
[image]

Can you please help me setupthis camera I just purchased from yourselves, errors all over show.

Thank you.

smudgy67, 2019-02-01:
Could you please tell me the order of loading firmware, win10, drivers, sdk, viewer, etc.
This is starting to drive me ting tong.
The SR300 camera was really simple. This has now been uninstalled.
I have the latest firmware for the D435i and updated this through the DFU tool (Intel RealSense D400 Series Signed Production Firmware v5_11_1).

Please help!?!

MartyG-RealSense, 2019-02-01:
Could you first try expanding open the 'RGB' section of the options in the RealSense Viewer and see if setting a lower resolution and / or FPS (such as 848x480 resolution and 15 FPS) reduces the occurrence of the USB overflow errors please?

[image]

dorodnic, 2019-02-03:
Thanks @MartyG-RealSense 

@smudgy67 - we will definitely try to make the process simpler. One concrete step I've been trying to get done is integrating the firmware update process into the Viewer. In general, unboxing the camera, plugging it to USB3 and downloading and running the Viewer should be the minimal sufficient set of steps. Calibration, UWP driver and firmware update are optional (while strongly recommended) 
There should be no conflict from the DCM.
 
Could you please specify what is the exact problem you are experiencing? 

The Overflows are an indication that the PC is not fetching all frames in time. Depending on the severity this can cause no streaming (when all frames are lost) or reduction in FPS. The same problem was most likely occurring on the SR300 as well (it simply lacks the mechanism to report it). 
Reducing FPS / resolution should take care of that.

smudgy67, 2019-02-11:
Hey MartyG and Dorodnic,

Thank you for your responses.

Finally back to the PC.

That's interesting, I will reinstall the SR300 camera then after I get the D435i to work properly.

As a matter of interest, do Intel provide longer leads for both the SR300 & D435i cameras. If not, what USB hub would you recommend, please? I have USB3.1, Type C ports.

I unloaded everything pertaining to the SR300 & D435i, and reinstalled the the latest SDK, this automatically installed the RealSense Viewer 2.18.1:
[image]

It appears everyting works fine here.

It's still no go with Mirillis Action and the background removal tool in relation to the D435i, I'm waiting for another support response from them. As previously mentioned I uninstalled the SR300. Are you saying that Action! is only supposed to be used with an SR300?
Also noticed that the D435i doesn't work with Windows Hello either.

Also, I'm having issues with Lightact 3.13, whereby it doesn't show the camera in the Stream list:
[image]

Also waiting for response from their support team.

I have followed your instructions as per install, however, am still having issues as your aware.
Please find attached the system information relating to my machine, it may be of some help.

Can you confirm that these are the latest drivers for the D435i, please?:
[image]

[image]

2006 seems very dated being 13 years old now.

I was kind of hoping the install process would just load all the bits to make it easier for the user. Must admit I have found the whole process a little tedious.

I do value your help though, second to none, thank you.

Kind regards,

Dominic

MartyG-RealSense, 2019-02-11:
Intel do not supply a longer official cable for RealSense, and using your own cable can provide variable results due to variation in cable quality.  My personal recommendation for cables proven to work with the 400 Series cameras is Newnex, who supply industrial grade equipment.  This includes fiber equipment to extend the camera up to 100 meters.

https://www.newnex.com/realsense-3d-camera-connectivity.php

You are correct that the developers of RealSense did not create Windows Hello support for the 400 Series.

I do not have knowledge of the workings of Action.  An alternative green screen tool compatible with the 400 Series is MixCast and its WildKey system.

https://mixcast.me/ditching-green-screen-mixcast-gdc-2018/?language=en_US

https://mixcast.me/docs/using-mixcast/?section=using-depth-sensing-cameras-5&language=en_US

The Windows driver dates are probably just referencing some generic part of Windows related to cameras.

smudgy67, 2019-02-12:
Hey MartyG,

Thank you for the link for the cables, will hang on for a while as these will cost $85 for 2, 2m cables.

So Windows Hello will be ok when I reinstall the SR300.

I loaded the MixCast though am having further troubles setting the camera and WildKey system up.

Please can you tell me exactly the versions of software I need to install the D435i camera correctly:
[image]

This page is referring to older versions than I have installed. Currently I have:
Firmware: 5.11.01.00
SDK: 2.18.1

On this occasion I haven't installed the calibration tool, however, I had calibrated previously before I unloaded all the software and started again.

Can you please put me straight on this for the correct install of the D435i?

This is starting to drive me ting tong now; I really thought it would have been an easy process to setup, a little disillusioned to say the least.

Though I'm sure we'll get it set up correctly in the end.

Kind regards,

Dominic

MartyG-RealSense, 2019-02-12:
Though Newnex is my personal recommendation for cables, a RealSense community member also recommended cables by OCP (I always remember the name because of the corporation in the movie Robocop!)

https://www.ocp.com/

I cannot guarantee that Windows Hello will work with the SR300.  It should be able to in theory, but it has a history of having problems.

I would not worry about the recommended firmware on that webpage.  They are recommending a version suitable for production applications (implying a need for stability), so it always makes sense not to recommend the very latest version until it is clear that unexpected problems with the new firmware are not going to arise.

For the D435i camera, the firmware should be a minimum of version 5.10.6.  So you are fine with having the newer 5.11.1 installed.  And you have the latest SDK.  So your setup of firmware and SDK looks fine.  

I do not have a D435i that I can run tests with, so one of the Intel staff on this forum such as Dorodnic or RealSense-Customer-Engineering will be better equipped to answer questions about setting it up after purchase.  

RealSense cameras are usually well calibrated when purchased and so calibration at the time of first use is optional.  In the case of the D435i though, the IMU component is not calibrated in the factory, so you will need to calibrate it yourself.  Intel have published an IMU calibration white-paper document to guide you in doing this.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf?language=en_US

Thanks for your patience!

smudgy67, 2019-02-12:
Hey MartyG,

Thank you for your patience too.

Lol, OCP, I remember the movie well, will take a look at that later, thank you.

So I'll leave the firmware as it's the latest version.
Lightact support just got back to me, and all their trials were done using the D435 camera.
They asked me to check the USB version:
[image]

As mentioned I have USB3.1 Type C, how would it be possible to acquire USB3.2? Is there an Intel update for this?

I will complete the IMU calibration and let you know the result.

MartyG-RealSense, 2019-02-12:
Thanks for your update.  I look forward to your results!  Unlike ED-209, you have more than 20 seconds to comply.  *smiles*

If your PC supports USB 3.2 and your cable is 3.1, the 3.2 standard can use the existing 3.1 cable that you already have.

smudgy67, 2019-02-12:
Hey MartyG,

Having nightmares doing the IMU Calibration.
You know this really sucks, now Python, really? And no windows IMU calibration facility?!?
I know this is not your fault but Intel, come on!

Ok, have Anaconda Python loaded on machine, this is from a course I'm currently doing in relation to OpenCV and its wonders, hmm, or not.

The version of Anaconda Python is 3.7.1, I tried following the first bit of the setup:
[image]

 and got this:
[image]

The second bit:
[image]

and got this:
[image]

I don't suppose there's one of you Dudes, who can just connect to my machine and perform this IMU calibration is there?

Majorly p*ssed off by it all! Sorry Dude, again not your fault.

smudgy67, 2019-02-12:
In relation to:
_Thanks for your update. I look forward to your results! Unlike ED-209, you have more than 20 seconds to comply. _smiles__
Lol, you know it. I wish I could turn the 'you have 20 seconds to comply' to this IMU calibration shite.

_If your PC supports USB 3.2 and your cable is 3.1, the 3.2 standard can use the existing 3.1 cable that you already have._
How would I know if it supports USB 3.2, please could you supply instructions as to how to find out, please?

lightact, 2019-02-12:
Hello,

I was just about to create a new issue when I found this one. It seems @smudgy67 has already posted the difficulties he is having in making D435i work with Lightact. 

The problem seems to lie in the following code. 

In Lightact code we have a _ScanDevices()_ function which essentially does this:
```
rs2::context ctx;
rs2::device_list devices = ctx.query_devices();
```
**query_devices()** is defined in _rs_context.hpp_ 
```
device_list query_devices() const
        {
            rs2_error* e = nullptr;
            std::shared_ptr<rs2_device_list> list(
                rs2_query_devices(_context.get(), &e),
                rs2_delete_device_list);
            error::handle(e);

            return device_list(list);
        }
```

This code detects D435 perfectly, but as @smudgy67 will gladly testify it doesn't seem to detect D435i. In his case, the device_list is empty. What can we do? Are there some library files (.h, .hpp...) that we should update perhaps?

The specs on our side are:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.18.1                          | 
| Language                            |  C++ | 
| Segment			|  others                          | 

Thanks!

smudgy67, 2019-02-12:
Look, all of you, I really appreciate your help on this and should we ever meet, the beers are on me =]

MartyG-RealSense, 2019-02-12:
@smudgy67   I do not have any experience with IMU calibration, so I will have to let the Intel guys answer that question, unfortunately.

@lightact   Thanks so much for your help with this case!

smudgy67, 2019-02-12:
Thank you @lightact for the help =]

@MartyG-RealSense , @dorodnic , @RealSense-Customer-Engineering, please can you help further in this issue? Thank you.

MartyG-RealSense, 2019-02-12:
I tried to think of a good way to explain USB 3.2, but the way it is implemented makes it a lot more complicated than it should be.

My understanding is that USB 3.2 is actually a new form of USB-C that gives faster data transmission through existing USB 3.1 cables.  And - this may be wrong - USB 3.2 support is added in the operating system (Windows 10 or Linux Kernel 4.18) so you do not need to buy a new computer to use it as long as your OS version supports it and the device attached to the USB port supports USB 3.2.

RealSense 400 Series cameras have been used successfully with USB 3.2 hubs, though the reports I have seen indicate that they can be more problematic than USB 3.1 hubs (for example, sometimes causing the camera's connection to be identified as USB 3.0 instead of USB 3.2).

lightact, 2019-02-12:
@MartyG-RealSense Could you please provide some feedback on my previous post about _query_devices()_ not finding D435i?

Thanks!

MartyG-RealSense, 2019-02-12:
I do not work for Intel, so there are some technical questions I have to defer to Intel support staff (who have access to more resources than I do) if I do not have knowledge on a subject or cannot find the answers through researching publicly available information.  I apologize for the delay in getting you an answer because of this - one of the Intel guys such as @dorodnic should be able to provide a reply later.

lightact, 2019-02-12:
ok, thanks for the reply. I'll wait for @dorodnic then.

smudgy67, 2019-02-12:
In response to:
_I tried to think of a good way to explain USB 3.2, but the way it is implemented makes it a lot more complicated than it should be._

_My understanding is that USB 3.2 is actually a new form of USB-C that gives faster data transmission through existing USB 3.1 cables. And - this may be wrong - USB 3.2 support is added in the operating system (Windows 10 or Linux Kernel 4.18) so you do not need to buy a new computer to use it as long as your OS version supports it and the device attached to the USB port supports USB 3.2._

_RealSense 400 Series cameras have been used successfully with USB 3.2 hubs, though the reports I have seen indicate that they can be more problematic than USB 3.1 hubs (for example, sometimes causing the camera's connection to be identified as USB 3.0 instead of USB 3.2).__

I'm just a little bemused that I have a nigh on up to date PC and the USB is set for 3.0:
[image]

Whereas Lightact have the USB set for 3.2:
[image]

The cable that came with the D435i is the standard 1m one, that should be good for up to 10Gbps, if I got the 2m, the speed would be halved.

Still I'm not getting why Lightact have USB3.2?!

Ok...
I have 4 USB 2.0 ports on the back of the PC and 2 on the front.
I also have 5 USB 3.1, Type C, ports on the back (One is marked as VR Ready) and 2 on the front.

Which would you recommend I use for the D435i, please?

Also in relation to the IMU calibration, is there any further help available, please?

Thank you.

dorodnic, 2019-02-12:
Hi @smudgy67 , @lightact 
Sorry for the delay. 

As I do work for Intel, I will do my best to assist. For the record, I'm not part of Intel support either, I am only managing this software project (librealsense) so there are topics beyond my control (firmware, calibration, site, schedules and more). librealsense issues section seem to be a convenient place for people to report and discuss problems, which is a good thing, so we are trying to help the best we can, but ultimately, RealSense does have official support channels (ping @RealSenseCustomerSupport)

@lightact - is the D435i detected using the RealSense Viewer? The reason for `device_list is empty` is one (or more) video or HID devices of the D435i not being present in Device Manager. 

@smudgy67 - anything above USB 3.0 should behave exactly the same.

The D435i was not designed to be a drop-in replacement of the SR300, I hope we do not advertise it this way and not confusing customers. The SR300 is designed to be, primarily, a smart desktop web-cam. As such it has Windows Hello and decent 3rd-party support for green-screen and teleconferencing. 
The D435i is designed primarily for robotics and drones and as such can operate outdoors, has IMU and is well integrated with the robotics operating system (ROS).

You don't have to go through the hurdles of calibration unless the initial factory calibration is not good enough for your needs. The device should be fully functional out of the box, and if I follow the conversation correctly it looks like it is. 

Let me know if you have any questions and sorry for all the confusion, we will try to improve

MartyG-RealSense, 2019-02-12:
Thanks so much for all that you do, Dorodnic.  I would answer everything on Intel Support myself if I could, but I inevitably have a few knowledge gaps (especially in the area of programming) that are covered by the engineers and developers such as yourself on the GitHub who work intimately deeply with the cameras every day.  

From the early days of the GitHub in the 400 Series era it was promoted as a place to ask questions, so it inadvertently became the "other" support forum when that may have not been its original intention when set up.  So I apologize for putting work on your shoulders!  :)

Regarding SR300 vs 400 Series and peoples' expectations, that was partially due to a shift in marketing.  the previous generation cameras were called Developer Kits, a term that was replaced by Depth Camera in the 400 Series generation.  So in the absence of the word Developer, purchasers came to expect the 400 Series to be more of a consumer plug-and-play product.

smudgy67, 2019-02-12:
Hey @dorodnic, @RealSenseCustomerSupport, @lightact 

No worries on the delay.

The D435i is detected using the RealSense Viewer V2.18.1.

Here is an image of my device manager:
[image]

and please find attached my system information: 
[ZEUS_12-02-2019_SystemInfo.txt](https://github.com/IntelRealSense/librealsense/files/2857184/ZEUS_12-02-2019_SystemInfo.txt)

From this data you'll see that I have USB 3.1, though it shows as USB 3.0, whereas Lightact's config shows USB 3.2 and they have no issues.

So is this the case where I need a D435 and not the D435i, as I thought the latter was the higher spec that would suit my applications such as Lightact and Mirillis Action, which is a live video recorder with RealSense technology, whereby it doesn't require a green screen.

So I have a SR300 which is back in the box. This worked with the Intel RealSense viewer, Windows Hello, however, wouldn't remove the background, at all! In addition this worked with some Lightact apps, however, not the latest UE4 project that I wanted to manipulate. Lightact use the D435 for this and have no issues at all with any of their projects.

I then ordered the D435i, thinking it was just the D435 with the addition of the IMU and a few other bells and whistles, of which I really don't need, though thought it would be pretty cool one day to use with a drone.

So the D435 and D435i aren't drop in replacements for the SR300, which as explained doesn't work properly with all the apps I want to use it with.

Though the D435i is what I have, so can please we get it up and running with your assistance, please?
I always thought that things were plug and play these days and this procedure would have taken minutes as opposed to countless days now,

I ran the calibration using the print-target-fixed-width.pdf document and this worked ok.
The IMU calibration on the other hand, I have real issues as previously mentioned and would really appreciate some help in doing this.

I'll tell you all now, I'll gladly order up a D435 if it will work with Lightact and Mirillis Action. The D435i doesn't! Which kind of really p's me off.
Marty G kindly gave me a link for MixCast that uses the D400 series cameras for background removal, though I can't get it to run with the D435i.
Though I'm sure you can understand my reluctance to get the D435, just in case that doesn't work.

So in conclusion, I really need your help here and am wanting to know:
- Are all the drivers correct and in their rightful place in device manager
- Why won't the D435i register in Lightact
- Why won't the D435i work for background removal in Mirillis Action or MixCast
- Why the IMU calibration is stalling and what can be done to complete based on the fact I have Anaconda Python 3.7.1 installed

You mentioned, 'The device should be fully functional out of the box'; I agree it works fine for the Intel RealSense... standard calibration V2.6.8.0 (not the IMU calibration), Viewer V2.8.1, Depth Quality Tool V2.8.1, however...
Not with LightAct, Mirillis Action or MixCast, which was the reason I purchased the camera in the first place.

Won't you please please help me... Thank you =]

smudgy67, 2019-02-13:
Can you please tell me why this case has been closed? Thank you.

smudgy67, 2019-02-13:
It appears I closed it:
[image]

Which was a silly thing to do and I apologise for doing it, please can the case remain open? Thank you.

MartyG-RealSense, 2019-02-13:
Underneath the box where you write a comment, next to the green Comment button, is there a button for you to open the discussion?

smudgy67, 2019-02-13:
Thank you MartyG for I've had two senior moments here, lol.

smudgy67, 2019-02-13:
Is there any further news on this case, please?

MartyG-RealSense, 2019-02-14:
As long as your case remains open, a member of Intel will provide a response.   I can't give a guarantee of when it will be though.  

Regarding your most recent question that you were seeking an answer to, I will add that the D435i is identical to the standard D435 except for the additional IMU.  So the D435i is only a higher-spec camera in that one regard.

The SR300 and the 400 Series use different camera technologies.   So although the RealSense SDK 2.i is programmed to work with the SR300, past applications (especially those made by non-Intel developers before the 400 Series was released) will not usually work with the 400 Series cameras.

RealSense tries to maintain compatibility with the previous generation of camera models where possible, whilst also not tying itself too tightly to past tech at the expense of progressing the product.  

For example, the original F200 camera model in 2014 used Windows 8 (and then Windows 8.1) instead of the far more widespread Windows 7, because Windows 8 had technology that could get the best out of the camera.  It caused a lot of concern at the time from users who wanted to use the camera with the Windows 7 they already had.  Likewise, the F200's requirement to use the most recent Intel Core processor caused problems for users with older hardware.

Regarding the 400 Series camera drivers ... because we are in an era where drivers download and install automatically, most users will never even see the drivers.   I haven't even looked myself, so I cannot say if they are in the right place.  If they are like the SR300 drivers, they might end up in slightly different places depending on the particular PC they are installed on.

The only reason I can think of why the D435i might not work with a product that does not need IMU functionality whilst the D435 does is that the product is allowing or denying a camera based on the camera's "PID" model identification code.  

I do not have an answer regarding why your IMU calibration is stalling.  Hopefully an Intel team member can assist you with that.

smudgy67, 2019-02-14:
Nice one MartyG, thank you for the detailed explanation.
I will wait for support to get back to me.

As previously mentioned, I am willing to purchase the D435:
https://click.intel.com/intel-realsense-depth-camera-d435-blue.html

This works with Lightact without issues so I'm going to order it up, then I'll have the holy trinity of intelligent webcams. Lightact kindly said if I send the webcams (SR300/D435i) they'll perform some tests on their end with them.

Please could you give me an appropriate contact number for customer support to discuss the my purchase of the D435 further? Thank you.

MartyG-RealSense, 2019-02-14:
There is no technical advantage in spending $10 extra on the Fifty Limited edition of the D435, as it is the same as the standard D435 camera except for a blue-finish casing.

RealSense technical support does not have a telephone number, as the support for RealSense is done through forums and online website tickets.  The Intel Click store does have a telephone number though.  Their website says:

"You may leave a phone message at this number: 877.649.5817 .   Dialing that number from outside the U.S. will not be toll-free.  We will get back to you within 2 business days".

smudgy67, 2019-02-14:
It's ok, $10 for blue finish, sounds cool and what's 10 bucks to the approx 380 I've spent already.
Really hope this one works?!?

Thank you for all your time spent helping out.

MartyG-RealSense, 2019-02-14:
You are very welcome!   Good luck!   :)

smudgy67, 2019-02-18:
Hey All,

The new D435 50th Edition came today and is so far working corectly with RealSense Viewer, however, it appears glitchy with the Depth Quality Tool.
In relation to Lightact, this appears to be working correctly too =].

I haven't as yet tried with Mirillis Action and MixCast, however, will keep you updated with progress.

Kind regards,

Dominic

MartyG-RealSense, 2019-02-18:
Thanks so much for your update - please keep reporting your results!

smudgy67, 2019-02-18:
I will, thank you.

In relation to the Depth Quality Tool, this sorted itself out following a restart.
It seems when lots of apps that are opened that use the D435, things start getting glitchy.

I've tried to set up MixCast though this is proving quite tedious, I will contact their support to discuss further. I'm guessing you need good light levels so will set up some lighting to suit. Will try the green screen I have too although that kind of defeats the object of the RealSense feature.

I would like to know what's happening re the help for the D435i, please; has there been any further update(s) on this, please?

MartyG-RealSense, 2019-02-18:
If you mean how you are seeking a support response from Intel staff members on this forum ...  as long as the case remains open then someone should get back to you on here, though I cannot offer an indication of when this will be, unfortunately.  Intel staff really do their best to answer cases, I promise.

smudgy67, 2019-02-18:
Ok MartyG, I will hang on and be most patient.
Getting the D435 to work will keep me quite for a while =]

ev-mp, 2019-04-23:
@smudgy67 , I'm sorry to bring this again, but can you outline the **currently standing issue(s)** that you experience with D435 ?
I've tried to "compile" the above comments into a single use-case, but I admit - failed miserably..
Please advise

smudgy67, 2019-04-23:
Hey ev-mp,

I uninstalled my D435i as no one could explain as to why it wasn't working with Lightact and Mirilis Action. I then purchased the D435 and although it works with Lightact, it doesn't do the background removal for Mirilis Action.

I do thank you for coming back on this query though, I thought you'd forgotten about me.

ev-mp, 2019-04-23:
Is there any SDK-specific issue that you can point out -  a broken demo or similar ?

ev-mp, 2019-04-24:
Both tools are closed-source apps, so the way to approach both vendors would be via their support channels:
https://mirillis.com/forum/
https://support.lightact-systems.com/report-a-bug/

At least in Lightact case they acknowledge that the tool was tested with D435 and may not work with D435i  (https://support.lightact-systems.com/realsense/)

Given this situation I'm inclined to close this issue for now unless there is a specific bug or feature request arise.

smudgy67, 2019-04-24:
Thank you for the links, had them anyway but thank you neither the less.

Lightact helped no end, and yes, they did state the tool was tested with D435; I thought in my, at the time wisdom, that the D435i would work with no issues, and that I could in the future use it for other applications that require the IMU feature.

That's fine ev-mp, to be honest, I thought it was forgotten about anyway.
Thank you for yours and Intel's revised interest in the issue though and assistance to date.

All the best.

ev-mp, 2019-04-24:
Thanks and good luck!
Sincerely.
@ev-mp

closed 2019-04-24 by ev-mp