realsenseai/librealsense #3199: ERROR!!  bash: ./<cpp file name> : No such file or directory

srnetsr, 2019-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.76                                         | 
| Platform                                 | intel aero compute board  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  cpp                      | 
| Segment			|                          | 

### Issue Description
question

Hi, I am trying to run tutorial code.
After finish installation, i ran the following codes
```
$ cd ~/librealsense/examples
$ cd ../realsense_camera/build
$ cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
$ make
$ sudo make install
```
Then I ran the cpp code, and the error was displayed.
```
$ ./_<cpp file name>_
bash: ./_<cpp file name>_: No such file or directory
```

Could you please help me to resolve this issue?

dorodnic, 2019-02-03:
Hi @srnetsr 
What version are you using? Also, could you please share the terminal log produced from `cmake` and `make`, as well as `ls` in `build`

srnetsr, 2019-02-06:
Hi @dorodnic , thank you for your response.

I installed librealsense by using the following command..
`$ git clone -b legacy --single-branch https://github.com/IntelRealSense/librealsense.git
`
In addition, here are what you asked.

Thank you for your help.

```
$ cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- Info: REALSENSE_VERSION_STRING=1.12.3
-- Building in a ROS environment
-- Using CATKIN_DEVEL_PREFIX: ~/librealsense/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: ~/librealsense/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- Configuring done
-- Generating done
-- Build files have been written to: ~/librealsense/build
aero_ubuntu@aero-ubuntu:~/librealsense/build$ make
[ 35%] Built target realsense
[ 38%] Built target cpp-tutorial-2-streams
[ 40%] Built target c-tutorial-3-pointcloud
[ 42%] Built target cpp-tutorial-3-pointcloud
[ 44%] Built target cpp-callback
[ 47%] Built target cpp-motion-module
[ 49%] Built target cpp-headless
[ 51%] Built target c-tutorial-1-depth
[ 53%] Built target cpp-alignimages
[ 56%] Built target cpp-callback-2
[ 58%] Built target cpp-tutorial-1-depth
Scanning dependencies of target find_params
[ 59%] Building CXX object examples/CMakeFiles/find_params.dir/find_params.cpp.o
[ 60%] Linking CXX executable ../devel/lib/librealsense/find_params
[ 60%] Built target find_params
[ 62%] Built target c-tutorial-2-streams
[ 65%] Built target cpp-stride
[ 67%] Built target cpp-enumerate-devices
[ 69%] Built target cpp-capture
[ 71%] Built target cpp-pointcloud
[ 74%] Built target cpp-config-ui
[ 76%] Built target cpp-multicam
[ 78%] Built target cpp-restart
[ 82%] Built target SR300-live-test
[ 86%] Built target R200-live-test
[ 88%] Built target F200-live-test
[ 93%] Built target ZR300-live-test
[ 95%] Built target offline-test
[100%] Built target LR200-live-test
```

```
~/librealsense/build$ sudo make install 
[ 35%] Built target realsense
[ 38%] Built target cpp-tutorial-2-streams
[ 40%] Built target c-tutorial-3-pointcloud
[ 42%] Built target cpp-tutorial-3-pointcloud
[ 44%] Built target cpp-callback
[ 47%] Built target cpp-motion-module
[ 49%] Built target cpp-headless
[ 51%] Built target c-tutorial-1-depth
[ 53%] Built target cpp-alignimages
[ 56%] Built target cpp-callback-2
[ 58%] Built target cpp-tutorial-1-depth
[ 60%] Built target find_params
[ 62%] Built target c-tutorial-2-streams
[ 65%] Built target cpp-stride
[ 67%] Built target cpp-enumerate-devices
[ 69%] Built target cpp-capture
[ 71%] Built target cpp-pointcloud
[ 74%] Built target cpp-config-ui
[ 76%] Built target cpp-multicam
[ 78%] Built target cpp-restart
[ 82%] Built target SR300-live-test
[ 86%] Built target R200-live-test
[ 88%] Built target F200-live-test
[ 93%] Built target ZR300-live-test
[ 95%] Built target offline-test
[100%] Built target LR200-live-test
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/_setup_util.py
-- Up-to-date: /usr/local/env.sh
-- Up-to-date: /usr/local/setup.bash
-- Up-to-date: /usr/local/setup.sh
-- Up-to-date: /usr/local/setup.zsh
-- Up-to-date: /usr/local/.rosinstall
-- Installing: /usr/local/lib/pkgconfig/librealsense.pc
-- Up-to-date: /usr/local/share/librealsense/cmake/librealsenseConfig.cmake
-- Up-to-date: /usr/local/share/librealsense/cmake/librealsenseConfig-version.cmake
-- Up-to-date: /usr/local/share/librealsense/package.xml
-- Up-to-date: /usr/local/lib/librealsense.so.1.12.3
-- Up-to-date: /usr/local/lib/librealsense.so.1
-- Up-to-date: /usr/local/lib/librealsense.so
-- Up-to-date: /usr/local/include/librealsense
-- Up-to-date: /usr/local/include/librealsense/rs.h
-- Up-to-date: /usr/local/include/librealsense/rs.hpp
-- Up-to-date: /usr/local/include/librealsense/rscore.hpp
-- Up-to-date: /usr/local/include/librealsense/rsutil.h
-- Up-to-date: /usr/local/bin/c-tutorial-2-streams
-- Up-to-date: /usr/local/bin/c-tutorial-3-pointcloud
-- Up-to-date: /usr/local/bin/cpp-tutorial-2-streams
-- Up-to-date: /usr/local/bin/cpp-tutorial-3-pointcloud
-- Up-to-date: /usr/local/bin/cpp-alignimages
-- Up-to-date: /usr/local/bin/cpp-callback-2
-- Up-to-date: /usr/local/bin/cpp-capture
-- Up-to-date: /usr/local/bin/cpp-config-ui
-- Up-to-date: /usr/local/bin/cpp-multicam
-- Up-to-date: /usr/local/bin/cpp-pointcloud
-- Up-to-date: /usr/local/bin/cpp-restart
-- Up-to-date: /usr/local/bin/cpp-stride
-- Up-to-date: /usr/local/bin/c-tutorial-1-depth
-- Up-to-date: /usr/local/bin/cpp-tutorial-1-depth
-- Up-to-date: /usr/local/bin/cpp-callback
-- Up-to-date: /usr/local/bin/cpp-enumerate-devices
-- Up-to-date: /usr/local/bin/cpp-headless
-- Up-to-date: /usr/local/bin/cpp-motion-module
```

```
:~/librealsense/build$ ls
catkin               cmake_uninstall.cmake  install_manifest.txt
catkin_generated     compile_commands.json  Makefile
CATKIN_IGNORE        CTestTestfile.cmake    test_results
CMakeCache.txt       devel                  unit-tests
CMakeFiles           examples
cmake_install.cmake  gtest
```

dorodnic, 2019-02-06:
Thanks, you can run the tutorials either using the short filename:
`cpp-tutorial-1-depth`
or local path `./examples/cpp-tutorial-1-depth`.
You can browse the code via `nano ../examples/cpp-tutorial-1-depth.cpp` or similar. 
Let me know if you have any questions

srnetsr, 2019-02-08:
Hi @dorodnic , thank you for your help.
The following command is working correctly.
`~/librealsense $ cpp-tutorial-1-depth`

However, in my case, I made cpp code and put it in the `librealsense/example` folder.
Also, I added the following lines in `CMakeLists.txt` in `librealsense/example`  folder.
```
# C/C++ tutorials and examples

add_executable(c-tutorial-1-depth c-tutorial-1-depth.c)
target_link_libraries(c-tutorial-1-depth ${DEPENDENCIES})

add_executable(cpp-tutorial-1-depth cpp-tutorial-1-depth.cpp)
target_link_libraries(cpp-tutorial-1-depth ${DEPENDENCIES})

add_executable(cpp-callback cpp-callback.cpp)
target_link_libraries(cpp-callback ${DEPENDENCIES})

add_executable(cpp-enumerate-devices cpp-enumerate-devices.cpp)
target_link_libraries(cpp-enumerate-devices ${DEPENDENCIES})

add_executable(cpp-headless cpp-headless.cpp)
target_link_libraries(cpp-headless ${DEPENDENCIES})

add_executable(cpp-motion-module cpp-motion-module.cpp)
target_link_libraries(cpp-motion-module ${DEPENDENCIES})

# added
add_executable(test_01 test_01.cpp)
target_link_libraries(test_01 ${DEPENDENCIES})
```

After that, I build it by using cmake, but it showed the results as the following.
```
~/librealsense$ text_01.cpp
test_01.cpp: command not found
```

dorodnic, 2019-02-10:
Hi @srnetsr 
Well, I'd recommend checking using `ls` what files were generated under `~/librealsense/build` and `~/librealsense/build/examples` and making sure the binary was actually created (maybe there was some error during compilation or cmake?)
With Linux, you cannot run local applications using just the name - `text_01` will not launch the program in current directory. To achieve this you should be calling `./text_01` or full path. 
When you write `text_01` without the prefix, Linux will look for the executable in the install folder, so it might work if you call `sudo make install`.

srnetsr, 2019-02-10:
sorry there were some typo at the end of my post.
I edited it. The file name is "test_01"
I will figure it out more.
Thnk you for your help, and I will ask a question later.

srnetsr, 2019-02-11:
Okay.
Let me summarized what I have done.
I hope it will help to resolve this issue....

1. Made test_01, and put it in the `librealsense/example`

2. Added two lines in `CMakeLists.txt` in `librealsense/example` as like shown in the previous post
```
# added
add_executable(test_01 test_01.cpp)
target_link_libraries(test_01 ${DEPENDENCIES})
```

3. Built
```
~/librealsense/build$ cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- Info: REALSENSE_VERSION_STRING=1.12.3
-- Building in a ROS environment
-- Using CATKIN_DEVEL_PREFIX: /librealsense/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /librealsense/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- Configuring done
-- Generating done
-- Build files have been written to: /librealsense/build
```

```
~/librealsense/build$ make
[ 35%] Built target realsense
Scanning dependencies of target test_01
[ 37%] Building CXX object examples/CMakeFiles/test_01.dir/test_01.cpp.o
[ 38%] Linking CXX executable ../devel/lib/librealsense/test_01
[ 38%] Built target test_01
[ 40%] Built target c-tutorial-2-streams
[ 42%] Built target cpp-tutorial-2-streams
[ 44%] Built target c-tutorial-3-pointcloud
[ 47%] Built target cpp-tutorial-3-pointcloud
[ 49%] Built target cpp-callback
[ 51%] Built target cpp-motion-module
[ 53%] Built target cpp-headless
[ 56%] Built target c-tutorial-1-depth
[ 58%] Built target cpp-alignimages
[ 60%] Built target cpp-callback-2
[ 62%] Built target cpp-stride
[ 65%] Built target cpp-tutorial-1-depth
[ 67%] Built target cpp-enumerate-devices
[ 69%] Built target cpp-capture
[ 71%] Built target cpp-pointcloud
[ 74%] Built target cpp-config-ui
[ 76%] Built target cpp-multicam
[ 78%] Built target cpp-restart
[ 82%] Built target SR300-live-test
[ 86%] Built target R200-live-test
[ 88%] Built target F200-live-test
[ 93%] Built target ZR300-live-test
[ 95%] Built target offline-test
[100%] Built target LR200-live-test
```

```
[... 77 lines cut ...]
* results of `ls`
```
~/librealsense/build/examples/CMakeFiles$ ls
CMakeDirectoryInformation.cmake  cpp-restart.dir
cpp-alignimages.dir              cpp-stride.dir
cpp-callback-2.dir               cpp-tutorial-1-depth.dir
cpp-callback.dir                 cpp-tutorial-2-streams.dir
cpp-capture.dir                  cpp-tutorial-3-pointcloud.dir
cpp-config-ui.dir                c-tutorial-1-depth.dir
cpp-enumerate-devices.dir        c-tutorial-2-streams.dir
cpp-headless.dir                 c-tutorial-3-pointcloud.dir
cpp-motion-module.dir            progress.marks
cpp-multicam.dir                 test_01.dir
cpp-pointcloud.dir
```

```
~/librealsense/build$ ls
catkin               cmake_uninstall.cmake  install_manifest.txt
catkin_generated     compile_commands.json  Makefile
CATKIN_IGNORE        CTestTestfile.cmake    test_results
CMakeCache.txt       devel                  unit-tests
CMakeFiles           examples
cmake_install.cmake  gtest
```

lramati, 2019-02-11:
As you can see from the the output of `sudo make install`, test_01 isn't
copied to `/use/bin/local/` and therefore isn't on the path, which is why
`$ test_01` fails.
You can add your executable to the `install()` directive in the
CMakeLists.txt file (line 123) to fix this.
Alternatively, you can rummage around in the build directory some more. I
would expect it to be in `librealsense/build/examples` like everything
else, although the line `[ 38%] Linking CXX executable
../devel/lib/librealsense/test_01` appears in your build log, so maybe try
`librealsense/devel/lib/librealsense`?

>

srnetsr, 2019-02-12:
Thank you for your help, @lramati !!
Finally, I found what was wrong with my setting.
I edited CMakeLists.txt as following you mentioned.

Thank you so much, @lramati 
Also, I thank @dorodnic  for helping me understand it.

closed 2019-02-12 by srnetsr