realsenseai/librealsense #3201: Align samples fail

mmoyes, 2019-02-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  n/a                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.17 & latest dev branch (2.18) }                          | 
| Language                            |  {C++/python }                          | 

### Issue Description
Trying to run align samples in either cpp (msvc 2017) or python. Trying both SDK 2.17 and compiled dev branch from github (2.18) (including pyrealsense2). I get nothing from the aligned depth thus end up with a grey image & black(c++) or blue(python/cv2) depth.

When I load in a bag instead (I tried d435i_walk_around.bag, d435i_walking.bag, stairs.bag) I get expected results.

#### Tried Samples
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/align-depth2color.py

https://github.com/IntelRealSense/librealsense/tree/development/examples/align

mmoyes, 2019-02-04:
Device is connected via USB 3.2

Problem is not specific to python as it also occurs in c++

RealSenseCustomerSupport, 2019-02-09:
----------------------------------------------

Hi @mmoyes,

As dorodnic suggested, please check if the camera enumerated as USB 2.0 using RealSense Viewer.
There is a slim chance that the cameras enumerated as USB2.0 even it is plugged in USB3.1 port.

mmoyes, 2019-02-10:
Tried again today on a new machine. On Ubuntu through USB 3.2 on librealsesne 2.18. Using the sample align code (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py). Same result.

`
Bus 002 Device 004: ID 8086:0b3a Intel Corp.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0b3a
  bcdDevice           50.b1
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 435i
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 435i
  iSerial                 3 850123050307
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         2819
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              110mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0
      iFunction               5 Intel(R) RealSense(TM) Depth Camera 435i Depth
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0
      iInterface              5 Intel(R) RealSense(TM) Depth Camera 435i Depth
      VideoControl Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength          119
        dwClockFrequency       48.000000MHz
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
[... 2079 lines cut ...]
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x1380  3x 896 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       7
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x1400  3x 1024 bytes
        bInterval               1
`

mmoyes, 2019-02-10:
[image]

Is the result from rs-align example

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

Hi @mmoyes,

Did you follow below instructions to install the prebuilt dkms package or have the patches by yourself? Suggest to install the prebuilt dkms and utils. And, run the prebuilt rs-align app and see how it works.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
$ sudo apt-get install librealsense2-dkms
$ sudo apt-get install librealsense2-utils

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

By the way, which firmware do you use? D435i requires FW version 5.10.13+.

mmoyes, 2019-02-25:
Hi @RealSenseCustomerSupport 
Yes This is on Windows and Ubuntu. Tried both with prebuilt and building from source. Also tested on multiple machines.
Firmware is 05.11.01.00 (I updated it when I got the initial problem)

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi @mmoyes,

Do you run other app like realsense-viewer, rs-capture and how it works with D435i?
You can enable librealsense log (in Win10, set LRS_LOG_LEVEL=DEBUG, in Ubuntu, export LRS_LOG_LEVEL=DEBUG) when you have the issue.

mmoyes, 2019-03-01:
Please find log attached.
[librealsense.log.txt](https://github.com/IntelRealSense/librealsense/files/2920801/librealsense.log.txt)

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @mmoyes,

It's probably something wrong about your D435i camera. Further clarification:
1. When you stream depth thru rs-align app, do you see the IR emitter is on?
2. Please use realsense-viewer to show us the left and right IR images.

mmoyes, 2019-03-04:
@RealSenseCustomerSupport 

Red IR emitter seems to be on.

[image]

What is the fastest way to get a replacement unit?

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi @mmoyes,

You can check with the retailer to have the replacement. Per the IR/depth/Color images, your D435i looks fine.
However, from the image of rs-align (running from the prebuilt app?), it looks the depth is all black, which is abnormal, unless no features there.

mmoyes, 2019-03-06:
Hi @RealSenseCustomerSupport 
Yep prebuilt rs-align. Here is the log

[2019-03-05-16_53_09.log.txt](https://github.com/IntelRealSense/librealsense/files/2933817/2019-03-05-16_53_09.log.txt)

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @mmoyes,

Not see the obvious error in the log. Does the IMU stream also normally by realsense-viewer or rs-capture? Suggest to try the new librealsense (2.19.0) and firmware (ex: 5_11_1_100) to see how it works?

davidbuckman, 2019-03-23:
I am having this same problem - did getting a new camera solve the issue?

mmoyes, 2019-03-25:
Hi @davidbuckman,
Sorry was waiting for my new unit to arrive (just did). Align sample works with new device. I also notice that out of the box the Motion module works a lot better too. My previous unit went crazy, then I manually calibrated it (as it is not factory calibrated) but looking at the non calibrated results of the new unit I reckon that was an issue too. 

An uneducated guess would be a gyro problem was causing depth frames to be delayed

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

Hi @davidbuckman,

You can consider to have a replacement or have a try of the dynamic calibration tool.

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Hi @davidbuckman,

Did you have a replacement already? If not, can you also install and try the dynamic calibration tool to dump the raw active calibration tables?
In Win10, the Intel.Realsense.CustomRW.exe tool in DC package under C:\CalibrationToolAPI\<version>\bin. (now, the version is 2.6.8.0)
In Ubuntu, the Intel.Realsense.CustomRW tool ins in /usr/bin.

Try to read raw calibration table from device and save into a binary file, supported table id 19, 1F, and 20, for example,
    Intel.Realsense.CustomRW -r -raw 19 -f mytable-19.bin
    Intel.Realsense.CustomRW -r -raw 1F -f mytable-1F.bin
    Intel.Realsense.CustomRW -r -raw 20 -f mytable-20.bin

ev-mp, 2019-05-14:
Linked to #3788

ev-mp, 2019-05-23:
@mmoyes  hello,
It seem that the issue has been resolved by replacing the unit, do you still need assistance on this?
Hi @davidbuckman, what is the current status ? - please update.

mmoyes, 2019-05-23:
Yep resolved for me. Feel free to close

davidbuckman, 2019-05-25:
Dynamic calibration did not work for us but we found a way to work around needing this functionality so are not planning on getting a new device.

ev-mp, 2019-05-27:
@mmoyes , thank you for confirmation.
@davidbuckman  - feel free to reopen in case you have a second thought.

closed 2019-05-27 by ev-mp