realsenseai/librealsense #3204: ModuleNotFoundError: No module named 'pybackend2'

bernland, 2019-02-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.18.0.515                           | 
| Language                            |  python                          | 
| Segment			|  AR                          | 

### Issue Description
ModuleNotFoundError: No module named 'pybackend2'

dorodnic, 2019-02-04:
`pybackend2` is not distributed via `pip install`, but you can build this module [manually from source](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source).
That said, `pybackend2` is rather an auxiliary library, that can be used to access low level UVC and HID devices, but not something you actually **need** to work with the camera.

closed 2019-02-05 by bernland

Dontla, 2019-10-02:
Hi, have you solved your problem, I have occured the same problem with you.

bernland, 2019-10-02:
No, I just skipped the example using `pybackend2`. Sorry!