realsenseai/librealsense #3209: D435i not supported via USB2 bus

neomanic, 2019-02-05:
We just received a few D435i devices, which we thought would be a drop-in replacement for a D435 in order to add an IMU to our system.

However, when plugged in via USB2, the D435i enumerates as an "Intel RealSense USB2", product id 0AD6. AFAICT this is the id for a basic 430 module. Neither the RGB sensor or the IMU is identified, only the infrared and depth stream. I can provide kernel logs and the output of rs-enumerate-devices, but there's no surprises in them.

Are there plans to support the D435i features over USB2?

| Required info | |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Up Squared  |
| SDK Version                            |  2.18.0                          | 
| Language                            |  n/a                          | 
| Segment			|  robotics                         | 

Steps to replicate: plug D435i into USB2 port.

RealSenseCustomerSupport, 2019-02-09:
----------------------------------------------

Hi neomanic,

Please connect camera USB 3.1 type C port to the cable first, then plug in the cable to PC. It should show up as USB3.1

Thanks,

dorodnic, 2019-02-10:
To put @RealSenseCustomerSupport response in context, yes, currently this is D435i's firmware limitation. Your point, however, is absolutely valid. 
I'm still checking to get at least official commitment for adding IR and RGB streams, similar to the D435. Adding IMU via USB2 would be more complicated.
Will update once I get a response.

neomanic, 2019-02-13:
Yes, I am aware it operates fine over USB3. I used another machine to check that it was a USB2 issue only before reporting here. 

We are not using USB3 on the Up board as we had nothing but trouble trying to get it to work in our robot's physical environment, and the data we need is fine over the bandwidth of USB2.

closed 2019-02-26 by RealSenseCustomerSupport

spk921, 2019-04-26:
@neomanic  so I can't get RGB stream with D435i ??? with USB2.0 port? Is it working with D435 then? 
@RealSenseCustomerSupport  Is there is any firmware update that supports USB 2.0?

neomanic, 2019-04-27:
@spk421 as far as I know there has not been a firmware update to support USB2 for the 435i. You can only access the depth stream over USB2, not color or IMU.

neomanic, 2019-04-27:
@spk921 as far as I know there has not been a firmware update to support USB2 for the 435i. You can only access the depth stream over USB2, not color or IMU.

spk921, 2019-04-27:
@neomanic thanks for the andwers. Then does D435 can acess color stream with USB2? Also, D435i doesn’t support MacOS currently but does D435 support MacOS?

neomanic, 2019-04-27:
Yes, with USB2 you can use depth and color streams at the same time under USB2. It works very well if you are happy with lower-resolution streams.

Sorry, while I develop from a Mac, my D435 code runs under Linux only so I don't know about MacOS support.