realsenseai/librealsense #3210: Ring-like error pattern in depth image of plane surface

alexanderhaertl, 2019-02-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  n.a.  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1    | 
| Language                            |  n.a. | 
| Segment			| Robot | 

### Issue Description
We experience a systematic error pattern in the depth image when we arrange the RealSense camera parallel to a large plane surface, see attached images. The error (resp. max. minus min. distance) is in the order of magnitude of one percent of the plane distance. The surface is a matte surface, for the images the emitter was enabled. 

We already verified that the error pattern is 
- independent of the particular camera (see attached image of other D415, we saw this on many different D415 cameras)
- independent on the firmware version used (the other camera has firmware version 05.11.01.00 installed)
- independent of the distance of the plane from the camera
- independent of whether the emitter is enabled or not
- independent of the parallelity of the plane to the camera (in the false color depth image, the error is, however, not as apparent as with a parallel surface)
- independent of the position above the surface (to verify that the surface itself does not show this irregularity, the second image "moving" was shot during slow movement of the robot arm parallel to the surface)

We also used the Depth Quality Tool, which shows that the RMS is completely within the expected bounds (approx. 0.28 % ). However, the regularity of the pattern is what makes us suspicious, and which makes it hard to use the camera for reliable measurements, compared to e.g. white noise in the depth image.

We verified that the intrinsic parameters of the camera are correct, by performing a manual camera calibration using MVTec Halcon with the rectified infrared images, which showed that the distortion of the rectified images is reasonably close to zero.

We verified that the epipolar lines are correct, i.e. chessboard corners have the same y-coordinate in the left and right infrared image.

We did not perform a field calibration of the camera so far, because it is not plausible that a misalignment of the infrared imagers lead to this error pattern (a misaligned camera must lead to a linearly distributed error among the depth image). Also it is not plausible that all D415 cameras we have are consistently miscalibrated.

Has anyone seen this kind of pattern as well? Is there an explanation for the source of this kind of error? How can we eliminate this systematic error?

## Steps to reproduce:
- take a D415 and a plane surface
- align the camera perfectly parallel to the surface
- watch the depth image in RealSense viewer

Thanks in advance!

[image]
[image]
[image]
[image]

vladsterz, 2019-02-05:
What a detailed post and good job being scientific about the origin of the problem, really great job. I've come across the same pattern in various ways. Once I tried to fit a a plane to the produced point cloud and the pattern is exactly the same. 
[image]

I suspected that the origin of the problem was radial distortion (radial pattern). 
[this](http://www.depthbiomechanics.co.uk/wp-content/uploads/downloads/2014/01/FINAL-3D-Body-Scanning-Paper-2013-checked.pdf) is the first random paper I came up google-ing **radial distortion depth images**. Since then I've not found the time to research into stereo pair undistortion, but I think it is still an open problem. A straightforward solution is to calibrate the sensors of the stereo pair as good as possible, but still slight distortion in the cameras would produce __wrong__ disparity -> depth inmho. 
If you can come up with something that helps correcting the produced depth, please share :D

agrunnet, 2019-02-05:
Yes what you have observed is correct. It stems from a mismatch between the lens distortion and the ability of the chip to correct for this type of distortion.  There is no fix to it unfortunately. However the D435 and D435i do not exhibit these fine ripples.

vladsterz, 2019-02-06:
@agrunnet have you looked deep into this problem? Can you share more detailed info about this problem, or redirect us to some info? Speaking of our application , we really prefer the D415 device for various reasons (narrow FoV, on chip HW sync among sensors). You state that the chip is not able to correct this kind of distortion, but can it be done in a post-process manner?

agrunnet, 2019-02-06:
Yes some companies have corrected this very fine depth structure by calibration each pixel vs distance to get a correction factor that they then apply and scale each point by but they had to measure at multiple distances. Also it is a per unit calibration correction meaning that cannot be generalized across different units unfortunately. It all stems from a residual aberration in the lenses that our chip rectification model cannot correct for.

vladsterz, 2019-02-06:
I've tried an similar approach, where I was trying to estimate the coefficients of a polynomial f(z) (1,2,3 degrees), over a dataset with planes, which corrected the depth. But then I stumbled upon a rather theoretical problem. Given a point (x,y,z) , changing its z coefficient should or should not change the remaining coefficients (x,y)?

TOMAxxHAWK, 2019-02-14:
I have the same problem with my D415 camera and it is driving me nuts.
A few thing I noticed:
- Moving the camera horizontally or vertically doens't change the pattern
- Changing the distance to plane changes the pattern a tiny bit
- It's independent of whether the laser is on or off
- Changing the (auto-)exposure or the gain doesn't change a thing

Based on these results and the locations of the "cirkles" I concluded that it might be caused by errors in the lenses. Maybe some sferical aberations might be the cause. There is a way to "correct" your data with a so called Fixed Pattern Noise Correction. Here is what you need to do:

1. Make sure your plane is as perpendicular to the z-axis as possible
2. Take around 50+ frames and take the average distance for every pixel over time, this will produce an average image
3. Devide every individual pixel value with the average of the entire image (this gives values around 0.8~1.2)
4. Store this data in a 1280x720 matrix (somewhere in text file for instance) or any format you can reconstruct

When you do a new measurement, you divide every pixel with the corresponding value in this calibration matrix. By doing so, you will remove the circular pattern. 
Note: when the plane is tilted you will notice that this matrix no longer functions as expected. I haven't figured a way to fix this yet.

alexanderhaertl, 2019-02-19:
Thank you all for your responses and support. It is good to know that this phenomenon is something others observe as well, and has nothing to do with our particular setup or improper use of the RealSense camera.
In fact, we also tried a calibration-based approach, similar to what @TOMAxxHAWK mentioned. However, we require a solution that does not require a per-device calibration. We figured that the lookup-table-based approach works well for our use case per device, but is not applicable to multiple devices using a single calibration.

TOMAxxHAWK, 2019-02-19:
I think that the RealSense is too sensitive that will cause these things to occur. So small impurities will cause these artifacts to show up. So every camera will have it's own errors. Unless Intel develops a tool to completely remove these problems, we are stuck with these artifacts. 

The main problem for me is that I am trying to determine what the ideal depth control settings are. But I am stuck because these artifacts really mess with my standard deviation of a plane. Because I want to measure a flat plane to get a feel for the accuracy of the camera without having to calibrate my data. I want my data to be as pure as possible, without having to filter or correct for certain artifacts.   

Something I would like to add is that this type of camera is designed to measure complex structures. It is not optimised for flat surfaces. When I try certain object the structures beat the spatial noise of the camera. If you want to use this camera for absolute distance measurements, a Time of Flight method is better.

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hi @alexanderhaertl,
As previously mentioned, the D4 chip is not able to correct this residual distortion for the lens.
We do, however, hope the responses from agrunnet, vladsterz and TOMAxxHAWK have provided some guidance/helps for you to address this issue in your application.
Do you have any further question regarding to this problem?
Thank you.

closed 2019-03-21 by RealSenseCustomerSupport