realsenseai/librealsense #3211: Building on unix from source leads to large file size

cansik, 2019-02-05:
I am currently building the librealsense shared libs on all platforms to release them with my application.  On Windows and MacOS I had no problems to do so and the shared library file size was about **~14 to 15mb**.

On Ubuntu, when building with `CMAKE_BUILD_TYPE=Release` and without any option turned `ON` except `BUILD_SHARED_LIBS`, the final binary file is about **~98mb**. With static lib building, the binary will increase to **~250mb**.

Checking out the **apt** version of the library for unix, the size is again about **~14 to 15mb**, which is what I would expect (same size for each os).

Is there a build flag I am missing or what could lead to such an tremendous filesize?

RealSenseCustomerSupport, 2019-02-16:
----------------------------------------------

Hi cansik,

To my understanding, the built .so file contains debug symbol and other stuffs.
You may try to strip out them bu "strip" command and see how it will look like.

Thanks!

cansik, 2019-02-16:
Great, that worked like a charm! Thank you very much!

closed 2019-02-16 by cansik