realsenseai/librealsense #3212: d435 gives Invalid Value in rs2_get_option error

moxo26, 2019-02-05:
| | |
|---|---|
|**librealsense**|2.18.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|828112072984|
|**Firmware Version**|05.11.01.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Please provide a description of the problem

Invalid Value in rs2_get_option(options:0x4bc7cf0, option:Emitter On Off):
hwmon command 0x7d failed. Error type: Invalid Command (-1).

I am trying to use camera with ROS Kinetic and I am assuming this same error is causing the ROS launch file to fail to connect to the camera aswell

dorodnic, 2019-02-06:
Hi @moxo26 
This is a known issue (see also #3191), will be resolved shortly. 
As a temporary work-around you can downgrade firmware or downgrade librealsense & ros-realsense combo.

dorodnic, 2019-02-10:
Resolved in [v2.18.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1)

closed 2019-02-10 by dorodnic