realsenseai/librealsense #3214: rs-convert unable to convert to PLY

HarshSharma12, 2019-02-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Win 10 Home  | 
| Platform                                 | PC|
| SDK Version                            |  2.18.0                          | 

### Issue Description

I recorded a video using the camera and when I try to extract the frames using rs-convert, it works as expected when trying to convert to PNG but if I specify the -l option to convert to PLY nothing happens. It executes for a few seconds and then exits without saving anything. Generated log files are also empty. Same behaviour is observed when trying to convert to BIN or CSV with the only difference being it outputs the number of frames processed but nothing is saved.

[image]

I already tried it by saving the file in different locations so it does not look like a permissions issue.

dorodnic, 2019-02-06:
I think that's a bug, we will look into it.

kogli, 2019-02-12:
+1. Having the exact same issue with conversion to bin and csv. My camera model is SR300.

amaanda, 2019-02-13:
I have the exact same issue using a D435. Converting to all the other formats works just fine.

yanite, 2019-03-07:
+1 no file

amaanda, 2019-03-07:
@dorodnic  I cloned the repo to a new folder and I still have the same issue with the ply conversion.

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

It should work as well as v2.19.0. Are we missing something?

    build$ rs-convert -i ./20190308_113002.bag -l out/
    100%
    PLY converter
     90  frame(s) processed

    build$ ls -l out/
    total 1153144
    -rw-r--r-- 1 12570346 Mar  8 11:32 _817.ply
    -rw-r--r-- 1 13009133 Mar  8 11:32 _818.ply
    .
    .
    -rw-r--r-- 1 13174367 Mar  8 11:35 _949.ply
    -rw-r--r-- 1 13668060 Mar  8 11:35 _950.ply

    build$ git log
    commit 43e30ddeccdfd01641e4da6e2748760dc472df01 (HEAD)
    Merge: 3dae1c41 7e91ffcc
    Author: Sergey Dorodnicov <sergey.dorodnicov@intel.com>
    Date:  Mon Mar 4 21:39:43 2019 +0200

        Merge pull request #3318 from dorodnic/bug_fixes

        Bug fixes

ynakayji, 2019-03-20:
@amaanda Do you record RGB as well as Depth ?
I had the same issue and have solved it by recording both.

amaanda, 2019-03-20:
@ynakayji no, I only had Depth and Infrared channels. Didn't think of that. Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

It could also work with depth and the Infrared channel if the infrared is outputting RGb8 format on the D415 camera.

closed 2019-03-27 by dorodnic

sophie-shin, 2019-04-12:
NO file +1
Jetson tx2 ubuntu 16 
viewer v2.16.0

```
nvidia@tegra-ubuntu:/usr/local/bin$ rs-convert -i '/home/nvidia/Documents/20190412_142928.bag' -l '/home/nvidia/Documents/ply.ply' 
100%
PLY converter

nvidia@tegra-ubuntu:/usr/local/bin$ rs-convert -i '/home/nvidia/Documents/20190412_142928.bag' -l out/
100%
PLY converter

```

flowb, 2019-10-12:
Hi, having the same issue and, I don't have the ability to recapture with RGB channels. Please help.

flowb, 2019-10-12:
Also, not sure why this issue was closed when no solution was ever found. 

If we need to have texture data in the .bag file in order to extract depth data, it's obviously a parsing issue. Telling people to adapt their workflows to sidestep the problems in the code isn't usually how we fix problems in software.