realsenseai/librealsense #3215: Align depth and color frames saved in PNG format

yuxiang-ricky-zhang, 2019-02-06:
--------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   { Linux (Ubuntu 16.04)   |   
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            |  {C++/python }                          | 
### Issue Description

Hi,

I have captured some depth and color images with the following code:

 rs2::frameset frames = pipeline.wait_for_frames();
rs2::depth_frame depth_frame = frames.get_depth_frame();
rs2::video_frame color_frame = frames.get_color_frame();
and save them both as png file using opencv imwrite().

Is there a way to align these depth and color images after they are saved in PNG format? I have studied the following examples:
https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp
https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb
but it appears in these examples that the alignment is done while the camera is streaming.

Thanks!
Any help will be appreciated.

dorodnic, 2019-02-06:
Hi @yuxiang-ricky-zhang 
This has been discussed several time on this forum. Once you have exported frames to PNG you have lost most of the data required to perform alignment. 
For the purpose of offline processing we offer [record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) example, allowing you to save raw data with all required calibration into ros-bag file format. 
Using this file format you can later feed it into any of the examples, including align.

closed 2019-02-06 by yuxiang-ricky-zhang

anug7, 2019-07-03:
@dorodnic , Can you tell what are the information needed to perform the offline alignment?

g138, 2019-08-22:
> Required Info
> Camera Model	D435
> Operating System & Version	{ Linux (Ubuntu 16.04)
> Platform	PC
> SDK Version	2.0
> Language	{C++/python }
> ### Issue Description
> Hi,
>
> I have captured some depth and color [...]

Hi 

I am also facing the problem of aligning depth and rgb images  saved separately, Can someone suggest some technique to solve out this problem.

cheers

anug7, 2019-08-22:
There's no api to align on image (after they are saved in image formats) levels in librealsense.. The alignment api works on frameset which has additional information (like scale paremeter, resolution, etc..) needed to align

So, you need to align the frames before saving them or same the frames in rosbag format (librealsense allows you to do that). And you can open the bag file and do what they are doing rs-align.cpp

g138, 2019-08-23:
Thank u for your reply.

When i am saving the image after aligning them there is loss and duplication of frames, how can i solve this problem?

cheers

anug7, 2019-08-23:
Yes.. I haven't found a way to solve the loss. I am not sure what you mean by duplication..There will be more RGB frames than depth frames. Once way to handle that would be storing the frames with timestamp as file names and lookup color frame corresponding to the depth frame, as the timestamp of RGB & depth frames are synchronized.

g138, 2019-08-23:
Thank u for your quick response
cheers