realsenseai/librealsense #3217: 4 cameras connected to PC by usb 3 hub are not working fine.

pratikshaandhare, 2019-02-06:
Hi,

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   {Ubuntu 16.04  | 
| Platform                                 | PC
| SDK Version                            |  { 2.17.0}                          | 
| Language                            |  {python }                          | 
| Segment			|  {3d reconstruction }                          | 

### Issue Description
 I am trying to capture point cloud from 4 cameras. I have modified code realsense_device_manager.py  as below. I have connected 4 realsense cameras to usb 3 hub and hub is directly connected to pc. 

` c = rs.config()
c.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
c.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 6)

self.device_manager = DeviceManager(rs.context(), c)
self.device_manager.enable_all_devices()

def enable_all_devices(self, enable_ir_emitter=False):

	print(str(len(self._available_devices)) + " devices have been found")
	self.filetobesave =  "file"
	i = 0
	for self.serial in self._available_devices:
	    # print("self.serial ", self.serial )

	    print("i old", i)
	    self.device = rs.context().query_devices()[i]
	    advnc_mode = rs.rs400_advanced_mode(self.device)
	    depth_table_control_group = advnc_mode.get_depth_table()
	    depth_table_control_group.disparityShift = 0
	    depth_table_control_group.depthClampMin = 0
	    depth_table_control_group.depthClampMax = 800
	    advnc_mode.set_depth_table(depth_table_control_group)
	    self.pc = rs.pointcloud()
	    self.decimate = rs.decimation_filter()
	    self.decimate.set_option(rs.option.filter_magnitude, 2 ** state.decimate)
	    self.colorizer = rs.colorizer()
	    self.enable_device(self.serial, enable_ir_emitter)
	    i = i+1
	    print("i update", i)

    def enable_device(self, device_serial, enable_ir_emitter):

        self.pipeline = rs.pipeline()

        # Enable the device
        self._config.enable_device(device_serial)
        pipeline_profile = self.pipeline.start(self._config)

        # Set the acquisition parameters
        sensor = pipeline_profile.get_device().first_depth_sensor()
        sensor.set_option(rs.option.emitter_enabled, 1 if enable_ir_emitter else 0)
        self._enabled_devices[device_serial] = (Device(self.pipeline, pipeline_profile))
        t_end = time.time() + 4 * 1

        while time.time() < t_end:
            if not state.paused:
                self.frames = self.pipeline.wait_for_frames()
                # self.frames = rs.context().query_devices()[i]

                self.depth_frame = self.frames.get_depth_frame()
                self.color_frame = self.frames.get_color_frame()

                # Grab new intrinsics (may be changed by decimation)
                depth_intrinsics = rs.video_stream_profile(
                    self.depth_frame.profile).get_intrinsics()
                # w, h = depth_intrinsics.width, depth_intrinsics.height

                self.depth_image = np.asanyarray(self.depth_frame.get_data())

                self.color_image = np.asanyarray(self.color_frame.get_data())
                

                depth_colormap = np.asanyarray(
                    self.colorizer.colorize(self.depth_frame).get_data())
                # state = AppState()
                if state.color:
                    mapped_frame, color_source = self.color_frame, self.color_image
                else:
                    mapped_frame, color_source = self.depth_frame, depth_colormap
                points = self.pc.calculate(self.depth_frame)
                self.pc.map_to(mapped_frame)
       
         
                filetobesave1 = self.filetobesave + device_serial + ".ply"
                points.export_to_ply(filetobesave1, self.color_frame)`

Always  it is showing 4 devices are detecting. but it is not able to capture 4 ply files. Sometimes it capture 2 ply files, sometimes 3. It is showing the error for3rd/4th cameras as below.

>     self.frames = self.pipeline.wait_for_frames()
RuntimeError: Frame didn't arrived within 5000

How can I resolve this? 

What is the best way to capture the ply files for 3d reconstruction using multiple cameras?
Kindly advice.

dorodnic, 2019-02-06:
We provide fairly comprehensive [multi-camera whitepaper](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf).
In practice there are a lot of system factors that affect multi-camera performance, there is no one solution.

pratikshaandhare, 2019-02-06:
Thanks a lot for reply. I will follow that. Is there any python script for multicamera?

dorodnic, 2019-02-06:
I think we don't have a multicamera python demo. You can take a look at:
[rs-multicam.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp)
Our python API is almost one-to-one with the C++ API, so it can give you direction. At glance, there is nothing obvious wrong with your script.

darrell-rg, 2019-02-21:
There is a multi-camera python demo here: 
[box_dimensioner_multicam](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam
)

It is currently broken on the main branch, use this PR to fix it: 
https://github.com/IntelRealSense/librealsense/pull/2949

As far as using more then one camera at a time,  a USB hub will give you problems, there is not enough bandwidth.  It is better that each camera gets its own USB controller, if your computer does not have enough built in you will have to add one or more  PCIe to USB3 cards.  I am using HighPoint  RU1344A card on linux and it seems to work fine.

closed 2019-02-26 by RealSenseCustomerSupport