realsenseai/librealsense #3220: API version mismatch: librealsense 2.16 required but 2.18 available realsense (ROS wrapper 2.1.4)

genessisperez, 2019-02-06:
Hi all,

I've been struggling for the last hour(s) with an API version mismatch, and I'm just trying to bring it here.

So, the ROS wrapper tag 2.1.4 has some last commits from today which require to have librealsense 2.16 instead of 2.18 and the compilation breaks at the end (error: no matching function, error: no type name, etc).

I reversed some of the changes (as yesterday everything was working after my pull) to the previous version and things are working again.

Changes I reversed:
CMakeLists.txt:
Line 36: find_package(realsense2 2.17.1 REQUIRED)
Line 88: add_dependencies(${PROJECT_NAME} ${PROJECT_NAME}_gencfg)

Now everything works again.

Please let me know if there's a cleaner solution to this problem than just reverting the commits from today.

Cheers,
Genessis

doronhi, 2019-02-07:
Hi,
The last commits on ROS wrapper 2.1.4 compiles well with librealsense 2.18.0, just as they are. 
Moreover, Line 36 in the CMakeLists.txt file is supposed to prevent you from trying to use librealsense 2.16.
It feels like I'm missing something here.
Could you add the problematic compilation log?

p.s.
It seems like this issue belongs with the ROS wrapper repository. Could you continue the discussion there?

genessisperez, 2019-02-07:
Hi,
(un)fortunately, I cannot reproduce the error anymore and the log file was overwritten (sorry, should have kept it). 
Anyway, I followed the same steps as yesterday and everything works without problems. So, it seems my changes were not the hacky solution but something got fixed (tried so many things yesterday, that now it's hard to say which one helped). Sorry I cannot help with more details.

Cheers,
Genessis

closed 2019-02-07 by genessisperez

doronhi, 2019-02-07:
Glad it worked anyway.