realsenseai/librealsense #3221: Miss aligned_frames between Color image and Depth image

RZz96, 2019-02-06:
*

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
color and depth frames are aligned but when i moved  object a bit faster then  they will not be aligned any more and specific points on color image will not be at the same pose in the depth image.

i appreciate your help

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@RZz9 Could you please provide below information?
1. Firmware version
2. librealsense/SDK version
3. Operation system(Linux/Windows or anything else?)
4. What's the resolution and FPS setting for color and depth?

RZz96, 2019-02-11:
im working with python in windows. 
using high resolution and (also i checked with 640*480) with 30 FPS.
pyrealsense2                       2.16.1.291
firmware : 5.08.15.00

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

@RZz96  The librealsense version you used is an old version. Please have a try with latest release such as v2.18.0 or v2.18.1 which supposed to fix this issue. Also please test with the Release build instead of Debug build.

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

@RZz96 One more question - are you using align-depth2color.py for the testing?

RZz96, 2019-02-12:
Thank you for the reply i upgraded the lib and still same problem  
To align : yes im using depth2color

RealSenseCustomerSupport, 2019-02-18:
----------------------------------------------

@RZz96 Now I also noticed the issue you mentioned for align-depth2color.py example. Then I tried rs-align in examples/align and found it's better. Could you please have a try with rs-align to see if it meets your expectation? For the issue of python example, will look into it further and update you later.

RZz96, 2019-02-18:
> @RZz96 Now I also noticed the issue you mentioned for align-depth2color.py example. Then I tried rs-align in examples/align and found it's better. Could you please have a try with rs-align to see if [...]

RZz96, 2019-02-18:
> example

i already uisng rs-align

closed 2019-02-18 by RZz96