realsenseai/librealsense #3222: Merge two realsense 415 cameras

codesian, 2019-02-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hi

We are trying to aling and combine in one stream, streams from two realsense 415 cameras, cameras are in vertical line trying to increase VFOV, with some overlap.  We read carefully Whitepapers and samples, (Multicam and Align, recomended in any issue). Align diferent streams seems easy, but we dont know how to do it playing with both cameras.

Thanks!

dorodnic, 2019-02-07:
Hi @codesian 
What output do you need from the camera? Are you looking for merged pointcloud or a "panorama-style" merged image?

codesian, 2019-02-07:
@dorodnic , thanks for your help. 

We need panorama-style images (RGB streams) to procesing it with openCV, and extract locations of this objects, but then we need to measure size of this objects (Depth streams). I dont know if we need pointcloud in order to align depth streams to get measurements.

codesian, 2019-02-19:
Hi! @dorodnic I try to go ahead with this problems.

First of all, in some threads ( [#1833](https://github.com/IntelRealSense/librealsense/issues/1833) ) i read that in order to align streams for multicamera configuration i can use rs2::align. But i dont know how and what is the expected result. maybe that way?

```
if (pipe.poll_for_frames(&frames))
		{
			if (pipe2.poll_for_frames(&frames2))
			{

				rs2::align align( rs2_stream::RS2_STREAM_DEPTH );
    			        aligned_frameset = align.process( frames );
 				aligned_frameset = align.process( frames2 );				
				std::cout << aligned_frameset.size() << std::endl;
			}
		}
```
As i can read `poll_for_frames` seems to be the better way to proceed in multicamera configuration.

To make panorama like images we try to make geometric transformations of two cameras pairs, (rectificate it and join) in order to get a merged image. (I dont know if its the right way)

Thanks

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi @codesian,

The rs2::align function is used to align depth and color pairs from each camera, instead of the images from different camera. You can try to stitch the color images using OpenCV. Since depth and color are aligned, you can get the correspondent depth value in pixel for the measurement.

closed 2019-04-08 by RealSenseCustomerSupport