realsenseai/librealsense #3223: LabVIEW Wrapper & D435: HelloWorld Color + Depth Example crashes

ryannazaretian, 2019-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / D435i | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Windows 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.17.1   | 
| Language                            |  LabVIEW                         | 
| Segment			|  Robotics                        | 

A while back I talked with @agrunnet about getting the HelloWorld examples working for LabVIEW. We're still unable to get the Color + Depth example to work. In fact, I cannot get the color channel to work at all from LabVIEW. 

I started to debug the examples a bit. Maybe I'm getting off in the weeds or maybe I'm figuring out the root cause. Hopefully someone can answer here. It seems the error pointer is getting ignored in several places, which seems to be causing full LabVIEW.exe crashes and also causes null profiles. So far I've debugged it to the rs2_video_stream_get_intrinics() call under the "rs3 configure all.vi". For the D435, the rs2_get_stream_profiles_count() call is returning 96. The rs2_video_stream_get_intrinics() function is failing on from=95, i.e. the last profile. It sets an error code that is never handled. Catching the error codes doesn't help much either. 

I setup a few probes around the config all function. You'll see that I've added the error handling so that I can try to track down the root cause for the empty profile (and maybe the crashing). 
[image]

Here are the probe values:
[image]

Like I said... the error isn't very useful, and it's really difficult to debug pointers from LabVIEW. I'm trying to get a debug build of the DLL setup w/ LabVIEW and Visual Studio 2017, attach, and debug that way, but I have so far been unsuccessful in getting that debug setup (not a Visual Studio user).

dorodnic, 2019-02-07:
Hi @ryannazaretian 
Thanks for sharing this info. @agrunnet please consider sharing the newer version of the sample.
I know that when the camera is in Advanced Mode, Y16 stream profiles of the stereo module come without firmware intrinsics, but some of the RAW color streams may also not be calibrated. Maybe this is the reason for the exception.

agrunnet, 2019-02-07:
Yes I will answer soon.

agrunnet, 2019-02-07:
It appears that you may be trying to get Color from the LEFT DEPTH IMAGER. This wont work for the D435, as it is monochrome. 
Let me clarify:
D415 has left and right imagers used for calculating depth from stereo. These imagers are actually color (wihtout IR-blocking filters). So they will see both visible and IR light. The sample we posted was to show the LEFT color imager of the D415. Note also that this color will be perfectly aligned with the depth, because depth is derived from it. 
In a same way you can view the LEFT imager of the D435, but it is MONOCHROME. So select the Y8 stream. 

Now, both of the depth cameras also have a THIRD imager, which is a 2MP color camera with an IR-block filter, and processed through an ISP to make it look better. 
You can access this imager by selecting RGB Ext. 
[image]
I will see about uploading a full Hello World for Third Color Imager

agrunnet, 2019-02-07:
Oh, and the capture proceed in the same way
[image]

ryannazaretian, 2019-02-12:
Latest help has unblocked us. Thanks!

closed 2019-02-12 by ryannazaretian