realsenseai/librealsense #3224: Change the RGB camera's exposure from inside the code

HamdiBouchech, 2019-02-07:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.09.14.00) | 
| Operating System & Version |   {Win (10) |                                        | 
| Platform                                 | PC |
| SDK Version                            |  { 2.14.1 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {others }                          | 

Hi, 
I was trying to change the exposure time of the RGB camera (D415 realsense) from inside my code, but i couldnt figure out how to do that. 
I could find how to do that for the IR camera using this function:
profile.get_device().first<rs2::depth_sensor>().set_option(RS2_OPTION_EXPOSURE, (float)(exposureMicroSec));

But i couldn't find the equivalent for the RGB camera ?

I tried doing:
std::string sensorName = sensor.get_info(RS2_CAMERA_INFO_NAME);
			if (sensorName.compare("RGB Camera") == 0) {
				sensor.set_option(RS2_OPTION_EXPOSURE, 5000.f);

				sensor.set_option(RS2_OPTION_GAIN, 27.f);
			}

which changes the RGB exposure inside the Viewer but inside the code that had no effect on the RGB colors, they are still coming dark.

Could you please tell me how to adjust the exposure time of the RGB camera within a c++ code ?

Thanks

MartyG-RealSense, 2019-02-07:
Before you set exposure manually, you should first turn off auto-exposure in your code.

Auto-exposure could be turned off with this instruction:

**sensor.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);**

 The link below may be of use to you in developing scripting that utilizes RS2_OPTION_EXPOSURE.

https://github.com/IntelRealSense/librealsense/issues/1663

HamdiBouchech, 2019-02-07:
yes i already turned the automatic exposure Off.
i used the mentioned api:
depth_sensor.set_option(RS2_OPTION_EXPOSURE, range.max)
to correctly set the exposure for the IR camera......but i would like to do the same thing for the RGB camera, is there something equivalent to:
auto depth_sensor = selected_device.firstrs2::depth_sensor();

to get the rgb_sensor instead of the depth_sensor ?

Thanks

MartyG-RealSense, 2019-02-07:
I believe that RS2_OPTION_EXPOSURE is the correct instruction, but there can be some variation on the initial **sensor.set** sensor setup definition.  For example, one user used the following code to set the RGB sensor exposure:

color_sensor.set_option(RS2_OPTION_EXPOSURE, 100);

HamdiBouchech, 2019-02-07:
How did he get defined the  **color_sensor** variable ?

I was experimenting with the realsense Viewer and i found the following strange behavior: 
when i turn on the rgb camera only i can set the exposure to get a nice images.
when the rgb camera is on and i turn the IR camera on as well, the rgb images become dark again and changing the rgb exposure has no effect
when Ir and RGB camera on, changing the IR exposure has effect on the RGB images, if i increase the IR exposure the RGB becomes brighter 
Does that mean we cannot change the exposure of RGB and IR at the same time and when the both cameras are on we can only adjust the exposure of the IR camera ?

Thanks

MartyG-RealSense, 2019-02-07:
Apologies for not quoting the source.  It is in the middle of a long script in the opening message of the discussion below:

https://github.com/IntelRealSense/librealsense/issues/2637

In regard to your other questions about different exposure effects: one of the Intel developer staff on this forum will be able to give a better answer than I can.

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

Thanks to MartyG-RealSense for all the information.

Hi HamdiBouchech,
Wonder with all the information MartyG-RealSense has shared, did you get it working?

Thanks!

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi HamdiBouchech,

Any updates on this? If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-03-14 by RealSenseCustomerSupport