realsenseai/librealsense #3225: Multi-camera field of video

MaryBD, 2019-02-07:
Hi. How much is the coverage of two IntelRealSense cameras that are placed perpendicular to each other? Assuming the cameras are on the ground how many degrees about them can they cover (if we look at them from the top view)? 
python, win 10, IntelRealSense d435

codesian, 2019-02-07:
Hi @MaryBD 

Maybe im wrong. In the [multicam realsense PDF](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper04.pdf), you've got exactly what you want. YHou can see here angle and camera overlaps...

If coverage means Horizontal field of view, you need to know distance to object, if realsense d435 have a 91.2 x 65.5 degrees.  Knowing the distance to the object with a little of trigonometry 

[image]

As you can see at 150 centimeters, you will have a theorical HFov of 306,34m and vFov arround 192,96

MaryBD, 2019-02-07:
Thanks for a quick reply. Do you measure and draw the above picture yourself? I cannot find it in the pdf file. So if it is 91.2 x 65.5 degree, yes that is what I am looking for. Do you get it from data sheet?

codesian, 2019-02-07:
You are welcome, yes i draw it for my project and share with you. Measurements on draw are based on theorical trigonometry and datasheet specs not on the real world measurements, just for orientation (They change a little, things are never like in datasheets :(  )

MaryBD, 2019-02-08:
Thanks for sharing then, and I agree with "things are never like in datasheets"!

closed 2019-02-26 by RealSenseCustomerSupport