realsenseai/librealsense #3229: D435i - RealSense Viewer: Could not open device. Last Error: Permission denied

FoxAviation, 2019-02-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | ?? | 
| Operating System & Version |   OpenSuse Leap 42.3 | 
| Kernel Version (Linux Only)    |  4.4.165                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.16.5 | 
| Language                            |  -/- | 
| Segment			|  Robotics | 

### Setup
1. D435i camera connected to PC via the USB 3 cable that was delivered with the camera
2. Opened ``realsense-viewer`` from ``librealsense v2.16.5``

### Error description
On connecting the camera, the RealSense Viewer shows the following error

``Backend in rs2_create_device(info_list:0x7fb4f000c530, index:0):``
``Could not open device. Last Error: Permission denied``

### Further investigation
- The same setup works fine for the D435 camera
- ``realsense-viewer`` in ``librealsense v2.10.2`` does not recongize the camera at all

Any advice is highly appreciated!
Thanks in advance

dorodnic, 2019-02-07:
Hi @FoxAviation 
[wiki/Troubleshooting-Q&A#q-i-ran-the-udev-rules-script-but-linux-still-get-permission-denied](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-i-ran-the-udev-rules-script-but-linux-still-get-permission-denied)
D435i did not exist when librealsense 2.10.2 was released.

closed 2019-02-17 by dorodnic

cychai1995, 2019-06-30:
sudo service systemd-udevd --full-restart