realsenseai/librealsense #3230: Segmenation fault when using realsense-viewer on odroid xu4 with D435 latest firmware, kernel patched as described.

florianspy, 2019-02-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When i start the realsense-viewer i first get the following output in terminal:
MESA-LOADER: failed to retrieve device information
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
But then the application opens.
As soon as i click on the icon to activate the camera ithe application closes and i get:
"Segmentation fault" displayed in console window

dorodnic, 2019-02-07:
Hi @florianspy 
Sorry to hear. 
Could you please share `dmesg` log after the crash?

antoniovillavicencio, 2019-02-08:
Have you tried setting 
```
cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false
```
before you run make?

dorodnic, 2019-02-10:
We did get very similar report from Nanopi-Neo4 user, but don't yet have this hardware locally. 
Any logs from the crash would help a lot and shorten the time to a solution.

florianspy, 2019-02-11:
This is the output of dmesg after the crash:
[  437.872744] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input16
[  438.880316] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  438.880345] uvcvideo: No streaming interface found for terminal 4.
[  438.880350] uvcvideo: No streaming interface found for terminal 5.
[  438.880359] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  438.880369] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[  438.880376] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  438.880381] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[  438.880805] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input17
[  439.849179] usb 4-1.2: usbfs: process 2802 (realsense-viewe) did not claim interface 1 before use

florianspy, 2019-02-11:
Hello with the tip from above cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false
realsense-viewer is not builded but i can at least call rs-color and see that data is coming in terminal.
Thank you alot for your help.
Is there another viewer available for the streamed data except for realsense-viewer?

antoniovillavicencio, 2019-02-11:
I've found some error in patch files. In `patch-arch.sh` on line 27, it can't find the link so I had to hardcode from https://.../patch-4.14.94.sign to ...patch-4.14.94.9.sign. Similar on `patch-realsense-ubuntu-odroid.sh` where it cant find .config folder under /linux-headers-...

Also, I reinstalled mali-x11 drivers with 
```
sudo apt-get install mali-x11 --reinstall
```

With this, realsense-viewer opens but it crashes when trying to display some image. It launches a warning of segmentation fault. I noticed that while compiling, x11 and mesa related packages launch several warnings all of the kind *ISO C forbids conversion of object pointer to function pointer* and similar.

florianspy, 2019-02-13:
On my raspberry pi after following the instructions given in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md i also get the segmenation fault as soon as i try to receive data from the realsense.

florianspy, 2019-02-13:
I applied your changes in the patch file of patch-realsense-ubuntu-odroid.sh and recompiled the librealsense now when i run it seems like the realsense-viewer is not build anymore.

antoniovillavicencio, 2019-02-13:
Yes, with `-DBUILD_GRAPHICAL_EXAMPLES=true` you are saying that you don't want realsense-viewer to build. Segmentation fault issue is an issue of realsense-viewer, not the libraries. This means you can still build your application and it will work. I couldn't run realsense-viewer yet but I'm working normally in ROS. I read something about Odroid having OpenGLES instead of OpenGL, I guess is the same with Raspberry but I don't know more :P

florianspy, 2019-02-14:
The problem is even the ROS driver seems to be broken by this problem i could not get any data. Instead i am getting the following exception:
[ERROR] [1549883789.649319258]: An exception has been thrown: hwmon command 0x7d failed. Error type: Invalid Command (-1).
when running roslaunch realsense2_camera rs_camera.launch

florianspy, 2019-02-14:
I find out that it is a problem with the latest firmware. 
I downgrade it to firmware version 5.10.13 and with that the ROS driver is working. So it seems like a bug in the latest firmware: 
https://downloadcenter.intel.com/download/28377/Latest-Firmware-for-Intel-RealSense-D400-Product-Family 5.11.1 
But realsense-viewer still result in the segmentation fault and the following output:
14/02 13:15:02,352 WARNING [2846728768] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]
 14/02 13:15:02,378 WARNING [2846728768] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @florianspy,

I don't observe the segmentation fault when opening realsense-viewer and having depth streams with D435 + 5.10.13.0 or 5.11.1.0. Let's check the correct installation and kernel patches applied first. You can try below patch which is working on my XU4 board with kernel 4.14.
https://github.com/freemanlo/librealsense/blob/odroid_build_4.14/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh
And, make sure to install Intel Realsense permission scripts located in librealsense source directory.
`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`
`sudo udevadm control --reload-rules && udevadm trigger`
And, make sure to install Intel Realsense permission scripts located in librealsense source directory.

Do you also try other librealsense app like rs-capture to see the segmentation fault?

florianspy, 2019-03-04:
I applied the kernel4.14 patch
I applied the permission script.
Still realsense-viewer gives me the segmentation fault, when trying to access the camera.
rs-capture does not give me the segmentation fault.

abylikhsanov, 2019-03-05:
@RealSenseCustomerSupport I can confirm that using the same script as yours did not really resolve the issues. I was able to connect D435i but sometimes I get an error that frames did not receive, here is the issue:
#3374

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi @florianspy,

Which exact board you use for this issue? you mean both Raspberry Pi 3 B+ and XU4 have the segmentation fault issue when trying depth stream in realsense-viewer?

Hi @abylikhsanov,

you use XU4, too? and, D435i firmware 5.11.10?

abylikhsanov, 2019-03-05:
@RealSenseCustomerSupport Yes, I use XU4 with kernel 4.14.102+ and 5.11.10 (The Latest) firmware.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @abylikhsanov,

Since your issue is due to hub power issue and you don't experience the segment fault when streaming using realsense-viewer, right?

Hi @florianspy,

It looks your issue only happens on the realsense-viewer. Suggest to try the latest realsense-viewer (ex: 2.19.0). And, please enable the log severity to Debug and output log to file in "settings->General" of realsense-viewer if the issue still there on your XU4.

abylikhsanov, 2019-03-07:
Hi,

I do experience the problem with the actual video streaming. I have
connected to the usb power hub but as soon as I start moving the robot base
I start getting to lose the frames again.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @abylikhsanov,

Since you mentioned this issue happens as soon as you move the robot base, I would think if the USB connection is not solid enough. Soft connection could affect the USB transfer quality/stability.
By the way, as I know, Newnex industrial grade passive USB 3.1 SuperSpeed cable is capable of connect Intel RealSense Camera series up to 5m (15 ft.) You can have a try.
http://www.newnex.com/realsense-3d-camera-connectivity.php
http://www.newnex.com/usb-3-1-locking-cables.php#USB3.1LockingAtoC

abylikhsanov, 2019-03-07:
Hi,

So would you say that the USB which came along with the camera is not good
enough?
I do not think that the problem is with the USB cable as if I connect the
camera to my laptop and move the robot base, I do not see any loss in the
connection.

My assumption goes to the kernel patch and not due to the power lack. From
the datasheet, the nominal required voltage for D435i is 5v which
odroid supplies through its USB port.

abylikhsanov, 2019-03-07:
Is it possible to use USB2.0 and have RGB and depth only? I do understand
that the performance would decrease but I do not really care as long as I
get a depth + rgb alongside. When I use USB 2.0 with the current setup, I
only get depth information and no RGB

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @abylikhsanov,

1. Need to further check the plan to support RGB and IMU streams for USB2.
2. Any failed kernel log and librealsense log when segment fault happens? or, you mean "No frames arrived" error in realsense-viewer?
3. Does rs-capture also have no this issue?
4. Please save your realsense-viewer (2.19.0?) settings to a json file. Like to see if it can be reproduced on my XU4 setup.

abylikhsanov, 2019-03-07:
@RealSenseCustomerSupport 
1. Ok
2. Yes, I mean the "Frames didn't arrive within 5 seconds" which happens to be randomly after which I am unable to receive any frames.
3. I don't think I have used it.
4. I am away from my main computer but I did save the json file earlier and the version was 2.18.1 with firmware version 05.11.01.00 (the latest version did not make any difference).

When I use on my laptop, I never get this problem. Also, I have used this patch:
https://github.com/freemanlo/librealsense/blob/odroid_build_4.14/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh

```
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

```

abylikhsanov, 2019-03-12:
@RealSenseCustomerSupport 
I have just tried the 5.11.4 firmware and 2.19.1 realsense but getting that problem. Which kernel patch did you use?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi @abylikhsanov,

I use the same patch.
https://github.com/freemanlo/librealsense/blob/odroid_build_4.14/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh
Please show me the full librealsense log (export LRS_LOG_LEVEL=DEBUG) and kernel log when starting moving the robot base and frame lost.
If you don't move the robot, frame lost will not happen?

abylikhsanov, 2019-03-14:
I also bought T265 and the frames get lost after 5-7 seconds. I have
decided to reinstall kernel and probably install the kernel 5.0 and see
what happens.

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi @abylikhsanov,

Any update? If no further question, will close this.

closed 2019-05-27 by RealSenseCustomerSupport