realsenseai/librealsense #3231: How to open .raw files from rs-convert?

amaanda, 2019-02-07:
I have tried a few different programs and also found some topics on MATLAB forums. Does the SDK have any support for this format?

dorodnic, 2019-02-07:
Hi @amaanda 
Well, raw is as it sounds, raw data, without headers. I have used [ImageJ](https://imagej.nih.gov/ij/download.html) in the past to open raw images, but the real purpose of this format is loading it via some API (matlab, python, C) and working with pixels directly.

amaanda, 2019-02-11:
I downloaded it and it works fine. Thanks.

amaanda, 2019-02-14:
@dorodnic I've been trying to load the file using `std::ifstream file("_Depth_92451.raw", std::ios::binary | std::ios::ate)` but apparently it's not working. How would you recommend to do so?

closed 2019-02-14 by dorodnic

amaanda, 2019-02-14:
@dorodnic do you have any recommendation on how to open these .raw files? I get different outputs when using ImageJ and C++ code.