realsenseai/librealsense #3234: Problems with rs-imu-calibration.py and possible fix

Dejauxvue, 2019-02-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 5.11.01.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            | 2.12.0.515| 
### Issue Description
Hi :)

I started working with the realsense D435i and wanted to calibrate the Accelerometer of the IMU. When using the tool rs-imu-calibration.py, I realized that some biases were off. E.g when placing the camera facing up the acceleration in Z direction was ~ 10.4 m/s^2 and when facing down, the acceleration was: 
[image]
 
[image]

When playing around with the calibration parameters, I realized that the bias for the y-coordinate is added with a negative sign, which resulted in the following measurement for bias_x = bias_y = bias_z = 1000: 
[image]

to cut the long story short: adding
        X[0,1] = -X[0,1]
        X[1,1] = -X[1,1]		
        X[2,1] = -X[2,1]
        X[3,0] = -X[3,0]
        X[3,1] = -X[3,1]
        X[3,2] = -X[3,2]
in line 575 fixed it for me.

My question is, is this a known issue and does a more optimal fix exist?
... or did I do something fundamentally wrong using the calibration tool?^^

thanks in advance

tim-fan, 2019-02-16:
Hello, I seem to have hit this issue also. 

After calibrating the device using rs-imu-calibration.py and committing the result to eeprom, I am still seeing norms up to 10.4 ms-2 when the device is stationary, in some orientations.

I tried the above suggested fix, this did bring my norms back much closer to 9.8, but it also negated the y-axis of the output imu values. 

I found I could get good norm values without negating the y axis by adding only these two lines to the script (again, adding at line 575):
```
X[3,0] = -X[3,0]
X[3,2] = -X[3,2]
```
(basically, negate the x and z bias terms)

But yes it seems like the underlying issue is in how the correction is applied - does this indicate a firmware bug?

Info about my setup:
```
Ubuntu 16.04.5 LTS
[ INFO] [1550322732.175372826]: Device Name: Intel RealSense D435I
[ INFO] [1550322726.872432988]: Running with LibRealSense v2.18.1
[ INFO] [1550322732.175463201]: Device FW version: 05.10.13.00
```

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Thanks for the feedback. This issue is under clarification. If any update, will let you know.

RealSenseCustomerSupport, 2019-08-21:
----------------------------------------------

Hi @tim-fan, @Dejauxvue,

Still have any trouble for this issue?

cichard, 2019-09-17:
I hav noticed the same behavior as you @tim-fan and @Dejauxvue. 
Moreover, I would like to understand one thing, maybe you can help guys ?
As you can see in your results, when facing down you get approximately -g on z axis. However it is stated in the [doc](https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/) that, IMUs reading are aligned to depth frame internally right ? In that way should not it be +g when facing down ? I am confusing myself maybe ?
Any help welcome 😁

Dejauxvue, 2019-09-27:
> 
> 
> Hi @tim-fan, @Dejauxvue,
> 
> Still have any trouble for this issue?

Hi,
yes, the problem seems to persist. We also tried it with newsest firmware and sdk on Ubuntu 16.

@cichard: unfortunately I don't have access to the device right now.. 
Are you sure that depth frames are alighned with z axis along the view direction?

RealSenseSupport, 2020-06-12:
Hi.

Will you be needing further help with this?

We believe the PR:

https://github.com/IntelRealSense/librealsense/pull/6029

addresses this issue.

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-07-06 by dorodnic