realsenseai/librealsense #3236: Red and Blue channel flipped

whg-res, 2019-02-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.18.0                          | 
| Language                            |  C/opencv                          | 

### Issue Description
I went through the install [instraction](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) and wrote a code by modifying Capture example (see following snippet). The problem is the captured images sometimes looks wrong. It seems that the Red and Blue channels are flipped. But it looks fine when I capture images in realsense-viewer. What am I doing wrong?

| OK                         |   NG                                                             |
|---------------------------------|------------------------------------------- |
| [image]  | [image] | 

    #include <iostream>
    #include <librealsense2/rs.hpp> 
    #include "cv-helpers.hpp"

    int main(int argc, char * argv[])
    {
        rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);
        rs2::pipeline pipe;
        pipe.start();

        while (1) 
        {
            rs2::frameset data = pipe.wait_for_frames();  
            cv::Mat img = frame_to_mat(data[1]);	
            cv::imshow("img", img);
            cv::waitKey(1);
        }
        return EXIT_SUCCESS;
    }

codesian, 2019-02-08:
Hi

I think that by default Realsense streams with BGR8, intel realsense viewer manage well this setups, but openCV didnt know. Force color mode on initial with:

```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 30);
rs2::pipeline pipe;
pipe.start(cfg);
```

Hope that helps!

whg-res, 2019-02-09:
Acturally, using RS2_FORMAT_RGB8 didn't worked for me. That still caused color problem when I enabled depth steram. But using RS2_FORMAT_BGR8 solved my problem. 

Thank you for your grate help!

closed 2019-02-09 by whg-res