realsenseai/librealsense #3237: d435i not detected on Mac OS Mojave on 2.2.18.1

moorage, 2019-02-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (not connecting) | 
| Operating System & Version |   MacOS Mojave 10.14.2  | 
| Kernel Version (Linux Only)    |  n/a                                        | 
| Platform                                 | MacBook Pro (Retina, 15-inch, Mid 2014) |
| SDK Version                            |  2.2.18.1                         | 
| Language                            |  realsense-viewer                          | 
| Segment			|  n/a                          | 

### Issue Description
I followed the compile instructions on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md for 2.2.18.1.  Everything compiled just fine in XCode.  When I run `realsense-viewer` and plugin the camera, it never registers as being plugged in.

I also tried running `rs-align` example, and got:
```rs-align 
RealSense error calling rs2_pipeline_start(pipe:0x7fd6e9454440):
    No device connected
``` 

When I tried the same thing on my linux box, on 2.2.18.0 (granted, not compiled from source), `realsense-viewer` works just fine.

kristjan-k, 2019-02-09:
I have the same problem.
The SR300 works fine but d435i is not detected.
It is showing on USB list.

ev-mp, 2019-02-12:
@moorage , at this time is D435i supported on  [Linux and Windows](https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1) platforms only.
#3122

leolodreamland, 2019-02-15:
how long til it is? i just bought two of these things!!! it said 435 on the product listing (amazon) but i've got 435i x2...

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

@leolodreamland Sorry about that! Currently please have a try on Linux and windows with D435i. Or please change to D435 for Mac.

leolodreamland, 2019-02-22:
did you have any clue on due date? or if it will appear?

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@leolodreamland Sorry, the schedule of D435i support on Mac is not very clear yet.  Please try D435i on Linux or windows first. Or you can try D435 if you don't need IMU.

leolodreamland, 2019-03-04:
Ok if I do use windows, is the d435i fully supportive of ms azure?

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

@leolodreamland  Sorry that I didn't have MS Azure PC in my hand to have a try. According to the release note, it said "D435i requires FW version 5.10.13+ and is currently supported on Linux and Windows 10 only. Windows 7, Android and Mac OS support under development to be included in future releases."  MS Azure is not included as supported OS.

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@leolodreamland Any other questions for this ticket?

leolodreamland, 2019-03-11:
No

closed 2019-03-20 by RealSenseCustomerSupport

keithahern, 2019-04-29:
FYI the version of librealsense bundled with Processing-realsense runs fine on Mojave with D435 so this is likely a regression.

https://github.com/cansik/realsense-processing

leolodreamland, 2019-04-29:
But a d435i?

keithahern, 2019-04-29:
Yes I mean a D345i