realsenseai/librealsense #3238: SDK 2.18.1 is not available in ubuntu 16.04 repo

rezaxdi, 2019-02-09:
Hi there,

I've update my d435 firmware to 5.11 and now i'm not able to use the ros wrapper. When SDK 2.18.1 will be available in 16.04 repo ?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.0                         | 
| Language                            |  ROS                          | 
| Segment			|  Robot                         |

ev-mp, 2019-02-10:
@rezaxdi , thanks for reporting.
There has been a hiccup with release server. The Debians are made public by now. 
Please check and confirm

rezaxdi, 2019-02-10:
Yup now it’s available.

closed 2019-02-10 by rezaxdi