realsenseai/librealsense #3239: D435i Calibration

barnjamin, 2019-02-09:
Does it make physical sense to combine the calibration results from raw IMU and corrected IMU from a subsequent run to achieve slightly better calibration of the IMU?

There is naturally some residual error after the first calibration and I'm hoping to eliminate as much of that error as possible.  I'm wondering if there is a physical basis for improving the IMU correction factors by re-running the calibration against corrected values.

If it isn't a silly idea how would I want to combine the sensitivity and bias estimates from raw calibration and calibrated calibration? Just element wise sum?

barnjamin, 2019-02-09:
Example data below, for the calibrated data I just commented out the line in the calibration script that turns off correction

raw.json
```
{
    "device_type": "D435i",
    "imus": [
        {
            "accelerometer": {
                "bias": [
                    -0.1928434669224553,
                    -0.27361064636689236,
                    0.007691682057752824
                ],
                "scale_and_alignment": [
                    1.0190453550818808,
                    -0.0017871002681981271,
                    0.003365105120994922,
                    0.010724323184422384,
                    1.0165117315943908,
                    0.027730820738040077,
                    -0.029201831950000645,
                    -0.046626023397331524,
                    1.01058429342749
                ]
            },
            "gyroscope": {
                "bias": [
                    -0.0009534871246483881,
                    2.6212704457687644e-05,
                    0.0002118860276996418
                ],
                "scale_and_alignment": [
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0
                ]
            }
        }
    ]
}
```

calibrated.json
```
{
    "device_type": "D435i",
    "imus": [
        {
            "accelerometer": {
                "bias": [
                    -0.3587258120768571,
                    -0.03136421391101966,
                    0.0917484808549185
                ],
                "scale_and_alignment": [
                    0.9993881402071313,
                    0.0021391227604891008,
                    0.007500704127766183,
                    0.018496037040335332,
                    0.9985925951891478,
                    0.059727445732979174,
                    0.00013898861483034654,
                    -0.06040434021278294,
                    0.9965238916857735
                ]
            },
            "gyroscope": {
                "bias": [
                    -0.0006235226957040305,
                    -0.0007241305967689304,
                    -6.125499022375056e-06
                ],
                "scale_and_alignment": [
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0
                ]
            }
        }
    ]
}
```

MartyG-RealSense, 2019-02-09:
Intel's white paper on D435i IMU calibration recommends that to improve the accuracy of the pose, a 3 axis level is used to  ensure the closest alignment to gravity.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf?language=en_US

I'm guessing that when it says a '3-axis level', the guide is referring to the 3 axes of the IMU's accelerometer.

[image]

barnjamin, 2019-02-09:
Understood.  I've worked to get it as close as possible and made the most level surface I could with what I've got on hand.  Just wondering if there is any way to iteratively improve the estimate.

MartyG-RealSense, 2019-02-09:
I see what the white-paper means now by the term '3 axis level'.  You learn something new every day!  :)

It does make logical sense to me to take a second reading to compare to the first to confirm the results.  It's a good scientific principle.  I can see though how it might be tricky to combine them.  For example, if one reading had some significant errors in it and you took an average of the two readings, the error would get incorporated into the averaged result.

As the guide says, it already tries to model the inaccuracies in the sensor and so take them into account.    It is hard to see what more you could do without a professional level device.  One of the Intel guys on this forum may have a suggestion though.

barnjamin, 2019-02-09:
https://hal-mines-paristech.archives-ouvertes.fr/hal-00473610/document this paper makes me think it may be possible, though their focus is on magnetometer calibration

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Hi @barnjamin,

Is there any further finding per your research? If no support required, will close this.

barnjamin, 2019-04-07:
I've not been digging into this, I was thinking someone smarter than me would take a look.

I've ordered a T265 to get pose information so I'll close this but I hope the SLAM for D435i is still in the works, I'd like to only require 1 camera for my product.

closed 2019-04-07 by barnjamin