realsenseai/librealsense #3241: Bag files and Matlab

TDVL, 2019-02-11:
Hello. I worked yesterday with RealSense Viewer in Windows and Matlab. I have extracted RGB, Infrared and depth images from *.bag file using Matlab. 
But how can I extract the XYZ coordinates of the scene and objects from *.bag file? I can't find this message in *.bag file. Where are stored these data? Also, how can I display the point cloud from the *.bag file using Matlab?
Thank you in advance for help.

MartyG-RealSense, 2019-02-11:
In a discussion on the Intel Support site about getting the points from a bag file in MATLAB, Intel support agent Alexandra posted a script.

https://forums.intel.com/s/question/0D70P0000069ICXSA2

lramati, 2019-02-11:
Also, if you didn't save a pointcloud object into the bag file while
recording, you'll need to construct it from the depth (and optionally
color) frames. The Matlab API is almost identical to the C++ and Python
APIs, so the examples from those languages should help you understand how
to do this with the `pointcloud` filter

TDVL, 2019-02-11:
Marty and Lior, thank you for the qiuck response. I stored the recording in *.bag file with the Viewer. Later, I used functions: bag, select, readMessages to access RGB, infrared and depth data in Matlab. Now, I don't know yet how to access the array with XYZ data and how can I access to the point cloud (for example with pcshow in Matlab)? These are my questions. I am a new RealSense user and I am not very familiar with ROS.

MartyG-RealSense, 2019-02-11:
I do not have a script for MATLAB point clouds based on a bag, but a community member called VHess did contribute a streaming example that may give useful pointers.  

https://forums.intel.com/s/question/0D70P000006ANoxSAG

lramati, 2019-02-11:
It sounds like you're using Matlab functions for accessing general bag
files. If you open it with the SDK instead, you'll get them as a simulated
camera, allowing you to use all of the SDKs functionality on the
pre-recorded data.

You can quickly modify the live pointcloud example by replacing line 8 with
```Matlab
cfg = realsense.config()
cfg.enable_device_from_file('file.bag')
profile = pipe.start(cfg)
```

TDVL, 2019-02-11:
And what about xyz array? Can I access, extract the array from Matlab command window? I need it for later image processing. I use the SDK Viewer only for recording.

lramati, 2019-02-11:
The pointcloud example ([found here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/pointcloud_example.m)) has a call to pcshow commented out on line 50. Is this what you mean?
If you need help setting up the SDK's Matlab wrapper, you can check the wrapper's [readme](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/readme.md)

TDVL, 2019-02-11:
I checked the example. But, can I in *.bag file simply access the XYZ array which contains the coordinates from Matlab?

lramati, 2019-02-11:
As far as I know, bag files recorded from the viewer do not contain XYZ
data in that format

TDVL, 2019-02-11:
OK, thank you Lior. Can I record data with SDK Viewer so that the data contain XYZ?

lramati, 2019-02-11:
I don't know that you can save pointclouds to a bag from the Viewer, but if
you record the depth stream into a bag, you'll be able to use the code in
the example from before to generate the XYZs. If you also record the color
stream into the same bag, you'll be able to generate UVs as well to color
the pointcloud

TDVL, 2019-02-11:
OK. What about 'ply' format? I think this format should contain the XYZ.

lramati, 2019-02-11:
It will, but last I checked the viewer didn't have a way to save PLYs for
every frame, and could only be triggered manually on individual frames

lramati, 2019-02-11:
You might be interested in the rs-convert command line tool, which can take
a bag with depth frames as input and output a PLY for each one

TDVL, 2019-02-11:
And how can I record than to have all the information (XYZ, RGB)?

TDVL, 2019-02-11:
I need an 640x480x3 image and it's xyz coordinates

lramati, 2019-02-11:
I would record a standard depth+color bag from the viewer, and then use the
SDK's pointcloud class to obtain XYZ+UV (color of point at x,y,z =
points(i) with texture coordinates u,v = tex_coords(i) is color(u,v))

TDVL, 2019-02-11:
Is there a Matlab version of this class?

lramati, 2019-02-11:
There is, in the Matlab wrapper I linked above. As the readme states, you
can install it either via the SDK's installer for Windows, or it can be
compiled from source for any platform

TDVL, 2019-02-11:
OK, so I need to record a bag file and then use one of the function from Matlab folder after SDK installation? Are there any specific settings?
Thank you for your guiding.

lramati, 2019-02-11:
Just make sure you're saving the depth stream in the bag. The
pointcloud_example I linked above should work as-is, but uses a live stream
from a connected camera. The 3 line modification I shared earlier will
update it to read from the specified bag file instead

TDVL, 2019-02-11:
OK, then I will save the bag file. After that, with your modifications in pointcloud_example function I can extract the RGB and XYZ arrays and I can also check the point cloud with pcshow. Am I correct? I hope yes.

lramati, 2019-02-11:
Yes. You'll need to uncomment line #21 to get the color frame, and
uncomment line #26 to generate texture coordinates though. You should be
able to understand how to access the texture coordinates just from the
example and pointcloud.m, but I don't know enough about Matlab to tell how
how to display a pointcloud with colored points

TDVL, 2019-02-12:
Hello Lior,
I tried with your modification, but it didn't work. I think there is a problem with lines #3 and #5, with the initial setup.

lramati, 2019-02-12:
What error are you getting? Did you follow the instructions in the readme?
There's a manual step you need to do after installing the wrapper from the
installer so that Matlab can find the library

TDVL, 2019-02-12:
Hello Lior,
I used the windows installer and than I copied the Matlab folder to another folder where the bag file was.
The error is: Undefined variable "realsense" or class "realsense.pipeline".
I think there should be some other settings too.

lramati, 2019-02-12:
You need to copy them still inside a folder called `+realsense`. This is
due to how Matlab defines packages.

TDVL, 2019-02-12:
I have copied everything from +realsense folder to another folder with bag file. Is this not correct?

lramati, 2019-02-12:
No, you need to copy the entire +realsense folder. The folder's name is
important to how Matlab works

TDVL, 2019-02-12:
I have copied the entire folder, but the new folder's name was not +realsense

TDVL, 2019-02-12:
I thing, the bag file should be inside the +realsense folder?

lramati, 2019-02-12:
So long as you pass the right argument to config.enable_device_from_file(),
the bag's location shouldn't matter

TDVL, 2019-02-12:
Now everything is in +realsense folder, the bag file and the Matlab files. I changed the line with you 3 lines. The error is:
realsense.pointcloud_example
Error using librealsense_mex
Failed to resolve request. Request to enable_device_from_file("20190212_082009.bag") was invalid, Reason: Failed to create
ros reader: Error opening file: 20190212_082009.bag

Error in realsense.pipeline/start (line 30)
                out = realsense.librealsense_mex('rs2::pipeline', 'start', this.objectHandle, config.objectHandle);

Error in realsense.pointcloud_example (line 11)
    profile = pipe.start(cfg);

TDVL, 2019-02-12:
This your code with my bag file:
cfg = realsense.config();
    cfg.enable_device_from_file('20190212_082009.bag');
    profile = pipe.start(cfg);

lramati, 2019-02-12:
It looks like it can't find the bag file. Try passing the full path
(C:\...\0000000000.bag)

TDVL, 2019-02-12:
Something is wrong, again the error:

realsense.pointcloud_example
Error using librealsense_mex
Failed to resolve request. Request to enable_device_from_file("D:\Work\Measurements\+relasense\20190212_082009.bag") was
invalid, Reason: Failed to create ros reader: Error opening file: D:\Work\Measurements\+relasense\20190212_082009.bag

Error in realsense.pipeline/start (line 30)
                out = realsense.librealsense_mex('rs2::pipeline', 'start', this.objectHandle, config.objectHandle);

Error in realsense.pointcloud_example (line 11)
    profile = pipe.start(cfg);

TDVL, 2019-02-12:
This is the code part now:

%profile = pipe.start();
    cfg = realsense.config();
    cfg.enable_device_from_file('D:\Work\Measurements\+relasense\20190212_082009.bag');
    profile = pipe.start(cfg);

lramati, 2019-02-12:
Looks like you've got a typo in your path there. Measurements+realsense
should probably be Measurements\+realsense ?

TDVL, 2019-02-12:
Now it is working (still), I got two empty figures and this is again the error message (but it is still busy):

realsense.pointcloud_example
realsense.pointcloud_example
Error using librealsense_mex
Failed to resolve request. Request to enable_device_from_file("D:\Work\Measurements\+relasense\20190212_082009.bag") was
invalid, Reason: Failed to create ros reader: Error opening file: D:\Work\Measurements\+relasense\20190212_082009.bag

Error in realsense.pipeline/start (line 30)
                out = realsense.librealsense_mex('rs2::pipeline', 'start', this.objectHandle, config.objectHandle);

Error in realsense.pointcloud_example (line 11)
    profile = pipe.start(cfg);

TDVL, 2019-02-12:
I am sure that the Matlab script is stucked, because it is still working.

TDVL, 2019-02-12:
It stopped now, the error message is:
Undefined function or variable 'Error'.

TDVL, 2019-02-14:
Hello Lior,
I changed the function a little, it is working now. Now, I will check that the function works well or not?
Is there a Matlab script that extract the RGB, INFRA, DEPTH IMAGE and the XYZ at once?

lramati, 2019-02-15:
You would need to record infrared from the viewer in order for it to be
readable from the bag, but assuming you did that,
pipeline.wait_for_frames() returns a realsense.frameset object containing
all your frames. It's API should be rather self explanatory, so you
shouldn't have a problem extracting each individual frame and putting them
in a list if that's what you mean.
Just remember that because of how the backend works, if you want to operate
on a significant number of framesets at once, you'll need to call the
keep() function on the frame objects so the backend can allocate more
memory for additional frames

TDVL, 2019-02-15:
Hi Lior,
Thank you for the suggestions.
I've recorded the RGB, infra and depth data in the bag file. I have many frames and I can extract the RGB, infra data for each frame separatelly, but the pointcloud_example calculates differently the XYZ data, I can't extract the coordinates for each frame separatelly, or I can (then I don't know how). Can I extract XYZ for each frame?
The pipeline.wait_for_frames() is in pipeline.m file, am I correct, or it is somewhere else?
I would like to extract the RGB, XYZ, INFRA and Depth with one function or script. Can I do this with the Matlab wrapper?

TDVL, 2019-02-17:
Hello Lior,
I tested the results for the obtained vertices, but they are not correct. Something is wrong. I have few frames, but I have only one array (307200x3)  with xyz values (not for all frames separatelly).

TDVL, 2019-02-21:
Hello Lior,
I tested the pointcloud_example function and I got a set of parameters and variables as result. One of them is pointcloud 1x1, but I can not open that variable to see the content. I assume that this variable should contain the RGB and XYZ arrays, like when I read in a PLY file.
What can be the problem?
Thank you in advance.

lramati, 2019-02-21:
The pointcloud object has functions on it you can use to access the actual
xyzuv values

TDVL, 2019-02-22:
Dear Lior,
I know that, but I can not access the pointcloud object and I don't know why. I made a screenshot of the Matlab window and I saved the help message. I attached them, you can see them.
After the execution of the code (the way you guide me to change the pointcloud_example function) I got the vertices and a pointcloud named object. I think this object should store the RGB and XYZ data of the point cloud, but I can't open it.
Can you help me with this? 
Thank you.
[image]
[im2.pdf](https://github.com/IntelRealSense/librealsense/files/2895715/im2.pdf)

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Hi TDVL,

Wonder if you have any update on this one?
Do you still need help? If so, would you please share out your code and clarify issues again?

Thanks!

TDVL, 2019-03-19:
Hello,
I have extracted the point cloud using the pointcloud_example.m function, but I can not extract the point clouds for each frame (like my friend in ROS). Is this possible with Matlab?
Thank you for the response.
Best regards,

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi TDVL,

If for extracting out PLY file, there is already a Matlab method created under points class:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/points.m#L15
You can utilize it to extract point clouds for a frame and save to a file. As it takes time to do so, you may need to do some handling.

Hope this can help.

Thanks!

TDVL, 2019-03-20:
Hello Support Team,
The PLY file is OK, I can, but I need speficic resolution, 640x480 or 1280x720. The PLY file doesn't provides these resolutions.
How can I extract each frame with the specified resolutions?
Thank you.
Best regards

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi TDVL,

Sorry I am not sure that I understand your question.
PLY files is for 3D, and the resolution is already determined in capturing time with stream configuration.

Thanks!

TDVL, 2019-03-21:
Dear Support Team,
Yes, PLY files are for 3D, but, when I export the scene in PLY format, the resolution is always reduced unpredictably. Hence, I won't get the desired resolution in PLY format. Naturally, PLY format contains data for all streams, but I need those data for the specified resolution.
Are there any solution for this (I mean to keep the resolution after exportation to PLY format)?
Thank you.
Best regards,

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi TDVL,

Maybe I missed something, what do you refer to by the resolution of a PLY format file?

Thanks!

TDVL, 2019-04-09:
Hi Support Team,
After I have exported the scene in PLY format, the resolution set in the Viewer is not preserved. It is easy to check, I think you have a RealSense camera to try it.
I need the point cloud with the specified resolution, the 1280x720, not the point cloud with the reduced resolution. 
Is this possible somehow? If not, I can't use this option saving the scene in PLY format.
Thank you in advance.
Best regards

RealSenseCustomerSupport, 2019-04-10:
----------------------------------------------

Hi TDVL,

Sorry that I didn't ask it clearly.
Yes, RS Viewer can also do the saving 3D stream to PLY files.
But I wanted to know what resolution information carried in a 3D PLY file that you are referring to and comparing to 2D resolution? And how you checked and had what kind of difference? Maybe you can share some screenshots or more info.

Thanks!

TDVL, 2019-04-15:
Hello Support Team,
The resolution is always lower in PLY format than in the preset. It is easy to check. Check the number of the points in the point cloud in PLY file and you will see that the resolution is reduced.
Thank you.
Kind regards

ev-mp, 2019-04-15:
@TDVL hello, when I read this
>After I have exported the scene in PLY format, the resolution set in the Viewer is not preserved. It is easy to check, I think you have a RealSense camera to try it.

I just thought - is it possible that you have the [decimation filter](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter) turned on when you export Pointcloud to PLY?

lramati, 2019-04-18:
Also keep in mind that points with no depth data aren't saved to the PLY
file, so this could also account for a loss of pixels

[1 more reply]

closed 2019-04-29 by RealSenseCustomerSupport