realsenseai/librealsense #3242: Failed to open scan_element (TX2 + D435i)

sophie-shin, 2019-02-12:
| | |
|---|---|
|**librealsense**|2.18.0 RELEASE|
|**OS**|Linux|

```
nvidia@tegra-ubuntu:/usr/local/bin$ ./realsense-viewer
 12/02 01:55:46,618 ERROR [548126666464] (types.h:214) Failed to open scan_element /sys/devices/3530000.xhci/usb2/2-1/2-1:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.1.auto/iio:device4/scan_elements/in_timestamp_en Last Error: Permission denied
 12/02 01:55:46,618 ERROR [548126666464] (backend-hid.cpp:962) Hid device is busy!
 12/02 01:55:46,619 WARNING [548126666464] (rs.cpp:273) null pointer passed for argument "device"
```

All sdk was working before installing ROS wrapper for realsense...

joanpepcompany, 2019-02-13:
The same problem with D435i in ROS and the realsense-viewer. 
**librealsense** 	2.17.1 
**OS** 	Ubuntu 16.04
**FW version on device**  5.11.1.0

On the other hand, D435 works perfectly on ROS and realsense-viewer.

ev-mp, 2019-02-18:
@joanpepcompany , the error indicate that the udev-rules for D435i are probably not in place.
The udev-rules are distributed either via `librealsense2-udev-rules` [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages), or with the SDK manual installation ([link](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)).

Please verify udev-rules and also the kernel patches to run D435i using either debian or manual installation paths, then use realsense-viewer for verification

sophie-shin, 2019-02-19:
Thank you all.
I recover Jetson tx2 again, and installed it.
Works fine now. Thank you.

closed 2019-02-19 by sophie-shin

joanpepcompany, 2019-02-25:
Solved using the first option, thank you.

228282416, 2019-05-17:
Hi~sophie-shin,How to install Intel D435i driver on TX2(JetPack4.2)?Could you help me?