realsenseai/librealsense #3243: RealSense R435 device will disconnect after plug-in for several minutes with ncs1

yechun1, 2019-02-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Ubuntu 18.04.2)   | 
| Kernel Version (Linux Only)    |  (4.15.0-45-generic)                                         | 
| Platform                                 | /NUC6i7KYB, BIOS KYSKLi70.86A.0037.2016.0603.1032 06/03/2016  |
| SDK Version                            |  { master }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Not sure if anyone else meet the same error, the RealSense device will disconnect after system running for a while when both plugin RealSense and Movidius. 

Reproduce:
1. plugin Movidius NCS1 and Realsense R435. Both devices detected by dmesg and lsusb
2. Do nothing and waiting for a while (did not run any realsense applications), 
3. After several minutes(about 3 minutes), check the device by lsusb,  Realsense device(8086:0b07) missed, and "kernel usb 2-2: USB disconnect, device number 5"`

dmesg:
> [ 1473.508216] usb 1-4: new high-speed USB device number 8 using xhci_hcd
> [ 1473.656684] usb 1-4: New USB device found, idVendor=03e7, idProduct=2150
> [ 1473.656686] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 1473.656687] usb 1-4: Product: Movidius MA2X5X
> [ 1473.656689] usb 1-4: Manufacturer: Movidius Ltd.
> [ 1473.656690] usb 1-4: SerialNumber: 03e72150
> [ 1477.932213] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
> [ 1477.953007] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
> [ 1477.953013] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 1477.953017] usb 2-2: Product: Intel(R) RealSense(TM) 435
> [ 1477.953020] usb 2-2: Manufacturer: Intel(R) RealSense(TM) 435
> [ 1477.953023] usb 2-2: SerialNumber: 821413020849
> [ 1477.954105] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
> [ 1477.954113] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
> [ 1477.954240] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
> [ 1477.955955] uvcvideo: Unable to create debugfs 2-5 directory.
> [ 1477.956062] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
> [ 1477.956066] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
> [ 1477.956070] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
> [ 1477.956073] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
> [ 1477.956323] input: Intel(R) RealSense(TM) 435: Int as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input19
> [ 1477.957223] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
> [ 1477.957235] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
> [ 1477.958088] uvcvideo: Unable to create debugfs 2-5 directory.
> [ 1477.958220] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
> [ 1477.958226] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
> [ 1477.958231] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
> [ 1759.232211] usb 2-2: USB disconnect, device number 5

/var/log/syslog

> Feb 12 10:39:55 yechun1 upowerd[1281]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4
> Feb 12 10:39:59 yechun1 kernel: [ 1473.508216] usb 1-4: new high-speed USB device number 8 using xhci_hcd
> Feb 12 10:39:59 yechun1 mtp-probe: checking bus 1, device 8: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4"
> Feb 12 10:39:59 yechun1 mtp-probe: bus: 1, device: 8 was not an MTP device
> Feb 12 10:39:59 yechun1 kernel: [ 1473.656684] usb 1-4: New USB device found, idVendor=03e7, idProduct=2150
> Feb 12 10:39:59 yechun1 kernel: [ 1473.656686] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Feb 12 10:39:59 yechun1 kernel: [ 1473.656687] usb 1-4: Product: Movidius MA2X5X
> Feb 12 10:39:59 yechun1 kernel: [ 1473.656689] usb 1-4: Manufacturer: Movidius Ltd.
> Feb 12 10:39:59 yechun1 kernel: [ 1473.656690] usb 1-4: SerialNumber: 03e72150
> Feb 12 10:39:59 yechun1 upowerd[1281]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4
> Feb 12 10:40:00 yechun1 colord-sane: io/hpmud/pp.c 627: unable to read device-id ret=-1
> Feb 12 10:40:04 yechun1 kernel: [ 1477.932213] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
> Feb 12 10:40:04 yechun1 kernel: [ 1477.953007] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
> Feb 12 10:40:04 yechun1 kernel: [ 1477.953013] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Feb 12 10:40:04 yechun1 kernel: [ 1477.953017] usb 2-2: Product: Intel(R) RealSense(TM) 435
> Feb 12 10:40:04 yechun1 kernel: [ 1477.953020] usb 2-2: Manufacturer: Intel(R) RealSense(TM) 435
> Feb 12 10:40:04 yechun1 kernel: [ 1477.953023] usb 2-2: SerialNumber: 821413020849
> Feb 12 10:40:04 yechun1 kernel: [ 1477.954105] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
> Feb 12 10:40:04 yechun1 kernel: [ 1477.954113] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[... 17 lines cut ...]
> Feb 12 10:40:04 yechun1 upowerd[1281]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.4
> Feb 12 10:40:04 yechun1 upowerd[1281]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.3
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) config/udev: Adding input device Intel(R) RealSense(TM) 435: Int (/dev/input/event19)
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Intel(R) RealSense(TM) 435: Int: Applying InputClass "libinput keyboard catchall"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) Using input driver 'libinput' for 'Intel(R) RealSense(TM) 435: Int'
> Feb 12 10:40:04 yechun1 upowerd[1281]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0
> Feb 12 10:40:04 yechun1 upowerd[1281]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) systemd-logind: got fd for /dev/input/event19 13:83 fd 62 paused 0
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Intel(R) RealSense(TM) 435: Int: always reports core events
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Option "Device" "/dev/input/event19"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Option "_source" "server/udev"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) event19 - Intel(R) RealSense(TM) 435: Int: is tagged by udev as: Keyboard
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) event19 - Intel(R) RealSense(TM) 435: Int: device is a keyboard
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) event19 - Intel(R) RealSense(TM) 435: Int: device removed
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input19/event19"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) XINPUT: Adding extended input device "Intel(R) RealSense(TM) 435: Int" (type: KEYBOARD, id 19)
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Option "xkb_model" "pc105"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (**) Option "xkb_layout" "us"
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (WW) Option "xkb_variant" requires a string value
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (WW) Option "xkb_options" requires a string value
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) event19 - Intel(R) RealSense(TM) 435: Int: is tagged by udev as: Keyboard
> Feb 12 10:40:04 yechun1 /usr/lib/gdm3/gdm-x-session[1140]: (II) event19 - Intel(R) RealSense(TM) 435: Int: device is a keyboard
> Feb 12 10:40:09 yechun1 colord-sane: io/hpmud/pp.c 627: unable to read device-id ret=-1
> Feb 12 10:41:00 yechun1 NetworkManager[843]: <info>  [1549939260.2560] connectivity: (eno1) timed out
> Feb 12 10:44:45 yechun1 kernel: [ 1759.232211] usb 2-2: USB disconnect, device number 5

yechun1, 2019-02-12:
Not sure the issue related with [usbids conflicts with Movidius Neural Compute Stick](https://github.com/IntelRealSense/librealsense/issues/2924)

dorodnic, 2019-02-12:
Hi @yechun1 
This is most likely not related to #2924. In that case, the NCS would be actively disconnected by the application layer, while here it seems the D400 is being disconnected by the OS. Do you think this be localized to a specific platform / bios?

mhkabir, 2019-02-17:
Observing this on UP Core (https://up-board.org/upcore/specifications/), but only in USB3 mode. In USB2 mode all seems fine.

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

@mhkabir
How many USB ports on your UP core?
For USB2 mode you said there's no problem. How did you connect for USB2 mode? Both D400 and NCS1 connected to USB2?
And how did you connect for USB3 mode? Do you see this disconnect issue if you only connect D435 to USB3?

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@mhkabir Could you please provide more information so that we can further look at this? Thanks!

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@mhkabir Are you still having the issue? Please provide more details so that we can look at it further. Thanks!

mhkabir, 2019-02-28:
How many USB ports on your UP core?
For USB2 mode you said there's no problem. How did you connect for USB2 mode? Both D400 and NCS1 connected to USB2?
And how did you connect for USB3 mode? Do you see this disconnect issue if you only connect D435 to USB3?

mhkabir, 2019-02-28:
> How many USB ports on your UP core?

1x USB3, 3x USB2 (via extension card)

> For USB2 mode you said there's no problem. How did you connect for USB2 mode? Both D400 and NCS1 connected to USB2?

Both connected to USB2. D435 connected to USB3 port using USB2 cable, "NCS1" (actually an Aaeon UP AI Core - https://up-board.org/ai-edge/ai-core/) is connected via USB2 on the PCIE port.

And how did you connect for USB3 mode? Do you see this disconnect issue if you only connect D435 to USB3?

D435 connected to USB3 port using USB3 cable (tried multiple cables). I did not test if it happens if only the D435 is connected to USB3.

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@mhkabir In this case, could you please try connecting D435 only or NCS1 only to USB3 port to see if any issue?

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@mhkabir Did you get chance to try connect D435 only or NCS1 only to USB3? Any update? Thanks!

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@mhkabir Do you still have issue? Waiting for your update. Thanks!

mhkabir, 2019-03-11:
Yes, I can confirm that with the AI core removed, the issue does not happen anymore.

closed 2019-03-18 by RealSenseCustomerSupport

mhkabir, 2019-03-18:
I'm not sure why this was closed - the primary problem still remains. With the NCS connected, the D435 is unusable in USB3 modes.