realsenseai/librealsense #3245: Cannot generate sln files for c# wrapper

dommy1C, 2019-02-12:
Hi
I downloaded the zipped file.
I installed cmake.
I get an error when running: cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_SHARED_LIBS=ON
[image]

dorodnic, 2019-02-12:
Hi @dommy1C 
Did you try with [CMake gui](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md)?

dommy1C, 2019-02-12:
No, I will try it now. Thanks

dommy1C, 2019-02-12:
Using it now. Get error. What directory do I actually select?
Thanks
[image]

dorodnic, 2019-02-12:
You should not use `C:\Program Files\...` as the build folder. Choose other location, you can also have it same as sources location. 
Also, if you are having trouble with CMake, [Intel.RealSense.SDK-2.18.1.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.18.1/Intel.RealSense.SDK-2.18.1.exe) includes the C# assembly.

dommy1C, 2019-02-12:
Ok, changed the target but now complains it cannot find Visual studio so looks like it does not support VS2017?

[image]

Also, i downloaded that other setup. I cannot see any C#wrappers? Can you please tell me the directory it should be installed in please?

Thanks

lramati, 2019-02-12:
If you look to the right of the configure and generate buttons, it says
"Current Generator: Visual Studio 16 2019". Try deleting the cache via the
dropdown menus and selecting Visual Studio 15 2017 as your generator instead

dommy1C, 2019-02-12:
OK, did all that.

Got new error:

[image]

lramati, 2019-02-12:
I'm not sure what to do about that one, sorry. In the meantime, don't
forget to select the BUILD_CSHARP_BINDINGS option at the top

dommy1C, 2019-02-12:
Ok - thanks. Seems to be a problem for other people as well.  Why it is so difficult to get sample code working for this I do not know. Looks like a decent camera as well. OK, will give up as my role waa to evaluate on behalf of my client.
Thanks all for trying to help :)

closed 2019-02-12 by dommy1C

dommy1C, 2019-02-12:
I solved this by installing Git. I guess I should be more careful when I look though the log files