realsenseai/librealsense #3247: Realsense Multicam on MacOS Mohave, pipeline fails to open more than one device.

punkstorm28, 2019-02-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |    MacOS 10.14.2  | 
| SDK Version                            |  2.18.1                          | 
| Language                            |  C++                           | 
| Segment			|  Multi Camera Setup                        | 

### Issue Description
Greetings,

I am using two RealSense D435 Cameras, they are connected to the MBP using a powered USB 3.0 Hub.
The program which derives from rs-multicam example works with one camera, but doesn't when another camera is connected.

The error is thrown when I call pipe.start(config), and the error is:

`RealSense error calling rs2_pipeline_start_with_config(pipe:0x7fcf3c028350, config:0x7fcf3c11ef20):
    Could not open device. Last Error: No such file or directory`

The code being:
~~~
 void enableDevice(rs2::device dev)
    {
        std::string serial_number(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::lock_guard<std::mutex> lock(_mutex);
        if (_devices.find(serial_number) != _devices.end())
        { return;  }
        if (platform_camera_name == dev.get_info(RS2_CAMERA_INFO_NAME))
        {return;}
        rs2::pipeline p;
        rs2::config c;
        c.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 15);
        c.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8, 15);
        c.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 15);
        c.enable_device(serial_number);
        try {
            rs2::pipeline_profile profile = p.start(c); //works for the first camera, fails subsequently.
            _devices.emplace(serial_number, view_port{ {}, p, profile });
        } catch (const rs2::error & e)
        {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
        }
        catch (const std::exception& e)
        {
            std::cerr << e.what() << std::endl;
        }
    }`
~~~
called at:
`        for (auto &&dev : ctx.query_devices())  {
            provider.enableDevice(dev);
        }
`

The interesting fact is, if I keep the second camera disconnected initially and connect it later when the first camera is already connected, the program works _sometimes_.

**More Information:**

I installed RealsenseSDK by building and installing manually, 
RealSense-viewer fails with `Segmentation fault: 11` 
and rs-multicam works when I connect one camera while starting it, then connect the other camera when it's already running, otherwise:
`RealSense error calling rs2_pipeline_start_with_config(pipe:0x7f9ac3799250, config:0x7f9ac356fd50):
    Could not open device. Last Error: Operation timed out
`

I have been trying different solutions like building RealSense SDK again, using a different hub, trying different configurations but I am not sure where the problem is,

Kindly help me understand what is wrong :)

dorodnic, 2019-02-12:
This is a bug / limitation, caused by libusb only allowing a single handle to a device. Should be addressed in future releases.

punkstorm28, 2019-02-14:
Thank you for the clarification, are we expecting it to be resolved from the libusb front first?

robbeadriaens, 2019-04-15:
Hi, I am having the same issue about not being able to connect multiple camera's. I'm also working on a MacOs Mojave and all the same parameters (Camera models, SDK version, Firmware version and language).  Is this issue already fixed or is there some way to work around this (by using other functions or libraries?)

vyomkeshj, 2019-04-18:
Hello @robbeadriaens, I don't think so, my workaround was switching to linux :)

equalsonics, 2019-06-22:
Hey all, was there any more updates to this issue for OS X? In terms of an update to the RealSense SDK or another lib that can be used to connect multiple cameras within OS X?

Thank you for your help1

moorage, 2019-07-25:
I noticed that by changing the order of serial numbers I send into `p.start(SERIAL_NUMBER)`, I was able to get 2 cameras working. Couldn't get more than that.

RealSenseSupport, 2021-01-06:
Please let us know if this issue still persists with the latest or recent build of the SDK. Thank you.

RealSenseSupport, 2021-01-15:
The issue should have been addressed in the recent SDK release. 
We will close this case unless otherwise. Thank you.

closed 2021-01-21 by RealSenseSupport