realsenseai/librealsense #3249: Rotate depth frame 90 degrees

Repulser, 2019-02-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Nvidia Jetson |
| SDK Version                            | 2.18.1                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
Hey, Im using the RealSense D435 mounted vertically on a robot since I need a vertical field of view.
I was wondering If it's possible to rotate the depth frame in python the same way I can rotate the RGB frame using OpenCV so that get_distance still works with the same coordinates.

Thanks!

vladsterz, 2019-02-13:
One way is to store the data from the depth frame to a cv mat and rotate it.

Repulser, 2019-02-15:
@vladsterz Thanks for the info! Can you provide an example?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi Repulser,

From what I can fin dout, if for rotating with open cv, there seems to be some methods already for the purpose:

https://github.com/opencv/opencv/blob/master/modules/core/include/opencv2/core.hpp#L1073
with supported rotation angles:
enum RotateFlags {
    ROTATE_90_CLOCKWISE = 0, //!<Rotate 90 degrees clockwise
    ROTATE_180 = 1, //!<Rotate 180 degrees clockwise
    ROTATE_90_COUNTERCLOCKWISE = 2, //!<Rotate 270 degrees clockwise
};

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi Repulser,

Do you still need help? If nothing else, it will be closed.

Thanks!

closed 2019-03-22 by Repulser