realsenseai/librealsense #3250: Device does not support AdvancedMode after Dispose

amatabuena, 2019-02-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.06 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  ---                                         | 
| Platform                                 | PC x86  |
| SDK Version                            |  2.17.0                          | 
| Language                            |  C#                          | 
| Segment			|  PC                          | 

### Issue Description
When stopping capture and release Pipeline and Config, if I try to initialize device in order to start capturing again (without reinitializing main program) I receive the error "Device does not support AdvancedMode" when trying to load json configuration again. Other times I receive "Access violation - no RTTI data!".

These errors are causing I can't stop and release camera and play it again.

Thanks.

ogoshen, 2019-02-14:
Could use some more info on reproducing this, but I suspect this is a double Dispose. 
In which case you can try and add this to the `AdvancedDevice`:
```cs
protected override void Dispose(bool disposing)
{
    // Intentionally empty, does not own the native device, only wraps it.
}
```

amatabuena, 2019-02-15:
Hi

You are right @ogoshen. Adding this Dispose override to AdvancedDevice solves the issue.

Thanks.

closed 2019-02-15 by amatabuena

ogoshen, 2019-02-17:
Great, thanks for verifying, it will be handled a PR soon.