realsenseai/librealsense #3251: Installation NOT reflected in kernel string, other apps don't find RealSense SDK

batchku, 2019-02-13:
Hello,

I have followed the instructions to install the SDK and dev tools.  
The installation completes without issues, but when I run:

```
modinfo uvcvideo | grep "version:"
```

I get:
```
version:        1.1.1
srcversion:     B768DE770525128619C215B
```

and I don't see the `realsense` string as the instructions indicate.

I'm able to run the `realsense-viewer` command and open the viewer app, but when installing other applications that requires the Realsense SDK to be installed, I get errors indicating that I don't have the SDK installed.

Anyone else see this issue?

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |4.18.0-15-generic  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { 2.18.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-02-13:
@batchku hello,

For the first part:
>modinfo uvcvideo | grep "version:"

Did you follow the [installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution) for kernel 4.16+ patching guide that would lead you [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)? If so - can you share the log obtained when running the required  `./scripts/patch-ubuntu-kernel-4.16.sh
` script ?

For the second part - Librealsense SDK uses a [standardized pkg-config interface](https://github.com/IntelRealSense/librealsense/blob/master/config/librealsense.pc.in) for inter-lib dependency resolution. 
Please provide the log error that you receive with the 3rd party software that you're using for review

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

@batchku Did you get it through now? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@batchku Any update?

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@batchku Any other questions about this ticket? If no, will close it accordingly. Thanks!

closed 2019-03-11 by RealSenseCustomerSupport