realsenseai/librealsense #3252: Autoexposure 'Mean Intensity Set Point' fixed at 1536 in Viewer

jkflying, 2019-02-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.13 & 5.11.1 | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  e.g. 4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description

In the BKM Tuning Guide one of the first items described is to adjust the exposure, eg. when autoexposure is enabled the `Mean Intensity Set Point`. However, this cannot be set from the Realsense Viewer, it is clamped to 1536 which is too bright for our application. I've tried a variety of versions of camera, firmware and USB. Note that the BKM shows a setpoint of 400, which isn't attainable. I'm not sure whether this is just a bug in the viewer, or whether it can't be set to anything else due to firmware restrictions/bugs/etc.

This was tested with a D435, D435i on both USB2 and USB3, with firmware versions 5.9.13 and 5.11.1.

dorodnic, 2019-02-13:
Hi @jkflying 
This is definitely a bug - the firmware is reporting an invalid range and the Viewer goes extra mile enforcing it.
If building from source is an option for you, a quick fix is to just comment-out [realsense-ui-advanced-mode.h:165-170](https://github.com/IntelRealSense/librealsense/blob/master/common/realsense-ui-advanced-mode.h#L165-L170)
We will try to include a proper solution in the next release.

jkflying, 2019-02-14:
Thanks for the workaround, that worked perfectly. Do you know if there are other places enforcing the invalid range, specifically the ROS interface? Will we be able to set the variable correctly there via a .json file, even if the firmware reports the wrong valid range?

dorodnic, 2019-03-06:
Resolved in [v2.19.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.19.0)

closed 2019-03-06 by dorodnic

Kaapeine, 2022-03-11:
Hi, can someone tell me how `Mean Intensity Set Point` is calculated and what its units are? I'm trying to find out the optimal value for my current use case. Thanks!