realsenseai/librealsense #3254: patch-realsense-ubuntu-lts.sh fails after qconf

ricber, 2019-02-13:
I'm trying to install the SDK on the Jetson TX2. The installation prcocedure [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](url) is not updated and do it's a little tricky to understand what are the lines to modify as there's no correspondence anymore. In this instructions you should refer to peaces of code than to lines that are specific to an older version of this file [https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts.sh](url)
Anyway I think I succeeded to modify it correctly but at some point the qconf window opens and if I close it the script crashes. This tool seems to be intended to configure the kernel to be installed but I do not know how to confirm the proposed configuration. 

This is the error I obtain:
[...]
hardware/nvidia/soc/t18x/kernel-dts/tegra186-soc/tegra186-soc-sata.dtsi
hardware/nvidia/soc/t18x/kernel-dts/tegra186-soc/tegra186-camera.dtsi
hardware/nvidia/soc/t18x/kernel-dts/tegra186-soc/tegra186-soc-mid-prod.dtsi
hardware/nvidia/soc/t18x/kernel-dts/tegra186-soc/tegra186-soc-actmon.dtsi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
qt5-default is already the newest version (5.5.1+dfsg-16ubuntu7.5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
scripts/kconfig/qconf  Kconfig

y
^Cscripts/kconfig/Makefile:22: recipe for target 'xconfig' failed
make[1]: *** [xconfig] Interrupt
Makefile:542: recipe for target 'xconfig' failed
make: *** [xconfig] Interrupt

|---------------------------------|------------------------------------------- |
| Operating System & Version |   Linux (Ubuntu 16 )                               | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                         | 
| Platform                                 |  NVIDIA Jetson TX2  |
| SDK Version                            |   2.18.1                          |

purplefishies, 2019-04-10:
What is the status of this bug ?  Not having the fixMakeFiles.sh  results in having drivers that do not get patched or built.

ricber, 2019-05-09:
No one have answered yet. Intel says that doesn’t officially support Jetson so they doesn’t care to this bug. I think this is stupid because Realsense devices produce a large amount of data that is common practice to process with a GPU. So, giving no support for the Jetson goes against their interests.

dorodnic, 2019-12-10:
We have updated installation_jetson.md and done basic testing on TX2 and Nano.

closed 2019-12-10 by dorodnic