realsenseai/librealsense #3256: Settings Deadlock after Cold Boot

mkehoe-ovio, 2019-02-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Win 10 | 
| SDK Version                            |  2.18.1                          | 
| Language                            |  C                           | 

### Issue Description
Calls to the SDK deadlock when setting certain settings.

Steps to reproduce:
1) Cold boot the system with the camera plugged in.
2) Before running the Intel real sense viewer, try running a program where it starts a pipeline and sets the auto exposure. Ex.
```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, cw, ch, RS2_FORMAT_RGB8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, dw, dh, RS2_FORMAT_Z16, 30);
auto profile = m_pipe.start(cfg);
auto device = profile.get_device();
for (rs2::sensor& sensor : device.query_sensors())
{
  sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1.0);
}
```
3) On the first set_option, the call will never return. 

Note that if the Intel realsense viewer is run first, or if the camera is unplugged an re-plugged in after the cold boot, the issue does not occur. Tested on multiple systems, on Windows only.

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

Hi @mkehoe-ovio,
Could you please try to call _rs::hardware_reset_() at the beginning of the program and see if this can temporary serve as a workaround for this issue?
Thank you.

mkehoe-ovio, 2019-02-20:
This doesn't seem to help the issue, only create more issues. If I call the hardware reset right before I create the pipeline, I can get one frame and then wait_for_frames never returns again. 

It seems that the hardware reset would be an async operation, and that the device would disappear and then reappear, yet I don't get any device changed notification from the context. 

Are you able to reproduce the original issue?

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

Hi @mkehoe-ovio,
The problem can be reproduced on my laptop very consistently and using the hardware_reset workaround has successfully brought the camera back to normal working state....

    rs2::context ctx;
    auto dev = ctx.query_devices().front(); // get first device
    dev.hardware_reset();
    rs2::device_hub hub(ctx);
    dev = hub.wait_for_device(); //wait for the reset and device to reconnect
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
     // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    auto profile = pipe.start(cfg);

mkehoe-ovio, 2019-02-20:
Thank you that works, I was using ctx.set_devices_changed_callback; using the device_hub.wait_for_device instead seems to work.

mkehoe-ovio, 2019-03-05:
Hi @RealSenseCustomerSupport 

When I use the above code, and I then call wait for frames, I never receive any frames. The exception what is "Frame didn't arrived within 5000". Here is an example.

```rs2::context ctx;
auto dev = ctx.query_devices().front();
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
rs2::pipeline pipe;
auto profile = pipe.start(cfg);

int i = 0;
while(i < 10)
{
  try
  {
	auto frameset = pipe.wait_for_frames();
	i++;
  }
  catch(rs2::error ex)
  {
	std::cout << "Wait for frame error " << ex.what() << std::endl;
	i++;
  }
}

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hi @mkehoe-ovio,
I was under the impression that it worked for you before. Any additional information would be very helpful.
We are actually now expecting a new firmware to fix this cold boot issue.
It will be available on the Intel RealSense once it is ready.
Thank you.

ryannazaretian, 2019-03-11:
Count me in on this issue. Been having issues getting my application (LabVIEW) to start. Seems as though if I run the RS Viewer first, then the LabVIEW code will execute. Otherwise, it hangs at startup. Like @mkehoe-ovio, I tried implementing a hardware reset, but the LabVIEW wrapper doesn't seem to like calling the initialization twice. I'll see if I can get the hub & wait_for_device() implemented.

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

The latest firmware 5.11.6.200 is available now.
Please download the firmware and the Windows/Linux firmware update tool via this link.
https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family
Please give it a try. Thank you.

smaesen, 2019-04-24:
I had the same issue. I can confirm that the latest firmware did fix the cold boot deadlock for me.

closed 2019-04-25 by RealSenseCustomerSupport