realsenseai/librealsense #3258: Camera intrinsics contradictory results

lelouedec, 2019-02-14:
Good morning, 
There is an issue with the app Intelrealsense.Customr -r to get intrinsics parameters of teh D435 and D415. 
The values given by this tool are completely wrong giving a wrongly aligned depth and RGB hence a wrong point cloud. 
While using the following piece of code the value for intrinsics parameters of the cameras are correct and very different from the one of the Customr tool. 

```
auto depth_stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
  auto color_stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
  rs2_extrinsics extrinsics = depth_stream.get_extrinsics_to(color_stream);
  std::cout << "Translation Vector : [" << extrinsics.translation[0] << "," << extrinsics.translation[1] << "," << extrinsics.translation[2] << "]\n";
  std::cout << "Rotation Matrix    : [" << extrinsics.rotation[0] << "," << extrinsics.rotation[3] << "," << extrinsics.rotation[6] << "]\n";
  std::cout << "                   : [" << extrinsics.rotation[1] << "," << extrinsics.rotation[4] << "," << extrinsics.rotation[7] << "]\n";
  std::cout << "                   : [" << extrinsics.rotation[2] << "," << extrinsics.rotation[5] << "," << extrinsics.rotation[8] << "]" << std::endl;

  auto resolution = std::make_pair(depth_stream.width(), depth_stream.height());
  auto i = depth_stream.get_intrinsics();

  std::cout << " Depth  INTRINSICS \n ";
  std::cout << "width : " + std::to_string(i.width) << std::endl;
  std::cout << "height : " + std::to_string(i.height) << std::endl;
  std::cout << "fx : " + std::to_string(i.fx) << std::endl;
  std::cout << "fy : " + std::to_string(i.fy) << std::endl;
  std::cout << "ppx : " + std::to_string(i.ppx) << std::endl;
  std::cout << "ppy : " + std::to_string(i.ppy) << std::endl;
  std::cout << "distortion coefficients : " << std::endl;
  std::cout << "model : " + std::to_string(i.model) << std::endl;

resolution = std::make_pair(color_stream.width(), color_stream.height());
  i = color_stream.get_intrinsics();

  std::cout << " Color INTRINSICS \n ";
  std::cout << "width : " + std::to_string(i.width) << std::endl;
  std::cout << "height : " + std::to_string(i.height) << std::endl;
  std::cout << "fx : " + std::to_string(i.fx) << std::endl;
  std::cout << "fy : " + std::to_string(i.fy) << std::endl;
  std::cout << "ppx : " + std::to_string(i.ppx) << std::endl;
  std::cout << "ppy : " + std::to_string(i.ppy) << std::endl;
  std::cout << "distortion coefficients : " << std::endl;
  std::cout << "model : " + std::to_string(i.model) << std::endl;

```
Using this C++ code is the only way to get a correctly aligned depth and a correct point cloud.

RealSenseCustomerSupport, 2019-02-24:
----------------------------------------------

Hi lelouedec,

Intel.Realsense.CustomRW tool with option '-r' is be used to dump out for calibration information for rectification. And Realsense SDK uses rectified images.
For your purpose, you would refer to the information dumped by "rs-enumerate-devices" tool with '-c' option.

Attached screenshots are from an example of D435 streaming 1270x720 depth+IR+color.
On the left side it shows the results from your code snippet, while right side showing results from "rs-enumerate-devices -c".
You can see the highlighted color and depth information from left & right side match.

Thanks!

Attachment(s):
D435_depth_color_1280x720_cali_info_for_RS-SDK.png - https://realsensesupport.intel.com/attachments/token/33Ef4V3ewyX3oyNRQyocPloeb/?name=D435_depth_color_1280x720_cali_info_for_RS-SDK.png

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi lelouedec,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-03-19 by RealSenseCustomerSupport