realsenseai/librealsense #3259: Stream volumetric data of D435 over network on unity

imanjj, 2019-02-14:
Hi,

I'm trying to stream D435 video and depth data over network. On the receiving side I'd like to see it in real time in Unity. Has anyone worked on sending the data efficiently over network? 
I'm new to realsense. There are many pieces that are confusing for me. 
1) To send data, I'm assuming everything should be the same as when we want to store data on disk. except that after encoding we send that data packet over network instead of saving it.

2) To receive, it should be like reading from disk except we are reading from the network port. 

Are my assumptions correct? 
I cannot find where this encoding, decoding is happening. Can someone please guide me?

Best,
Iman  

Camera Model: D435
Firmware version: 05.10.03.00
OS: Win 10
Platform: PC
Language: Unity

dorodnic, 2019-02-14:
Hi @imanjj 
There is this project - https://github.com/krejov100/EtherSense
AFAIK it is still work-in-progress, but you might find it useful.

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hi @imanjj,

How is everything going based on below GitHub project?

closed 2019-04-09 by RealSenseCustomerSupport