realsenseai/librealsense #3260: How can i build the python wrapper from source in mac os environment

syh4661, 2019-02-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/python }                          | 
| Segment			|  {Robot/}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I  build all the librealsense package in xcode and now i want to use

pytho wrapper but here comes out 

No UVC backend selected. Please #define exactly one of RS2_USE_LIBUVC_BACKEND, RS2_USE_WMF_BACKEND, RS2_USE_WINUSB_UVC_BACKEND or RS2_USE_V4L2_BACKEND

error when i build in wrapper/python folder build

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

Hi syh4661,

For building python wrapper off Mac OSX, you can try following, which works for me off High Sierra:

1. Follow the instructions for building off Mac OS and add in python related cmake options
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

Something like this (using python2 case):
_cmake ../ -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/Library/Frameworks/Python.framework/Versions/2.7/bin/python2_

2. And then copy over the built .dylib and .so files to be next to your ppython code
a) realsense .dylib files (under build=>Debug)
b) realsense python .so files (under build=>wrapper=>python=>Debug)

Thanks!

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi syh4661,

Wonder if there is any update? Did it work for you?

Thanks!

smanthe, 2019-03-03:
For me it worked with Python 3.7 on a Mac.

Nevertheless it would be nice to be able to install librealsense2 with pip3 for Python 3.7 on Mac. Maybe only the setup.py has to be expanded?!

I'm always getting the following error when trying to install librealsense with pip and Python3.7 on Mac:
```
~/$ pip3 install --user  librealsense2
Collecting librealsense2
  Could not find a version that satisfies the requirement librealsense2 (from versions: )
No matching distribution found for librealsense2
```

Thank you

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi smanthe,

Thanks for the update. Glad to know that python3.7 works off your Mac.
Unfortunately pip installation off Mac is not ready yet. Please check back later for more python support on Mac.

I am going to close this one if nothing else.

Thanks!

closed 2019-03-14 by RealSenseCustomerSupport

whubaichuan, 2020-09-10:
> Hi syh4661,
> 
> For building python wrapper off Mac OSX, you can try following, which works for me off High Sierra:
> 
> 1. Follow the instructions for building off Mac OS and add in python related cmake options
>    https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
> 
> Something like this (using python2 case):
> _cmake ../ -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/Library/Frameworks/Python.framework/Versions/2.7/bin/python2_
> 
> 1. And then copy over the built .dylib and .so files to be next to your ppython code
>    a) realsense .dylib files (under build=>Debug)
>    b) realsense python .so files (under build=>wrapper=>python=>Debug)
> 
> Thanks!

@RealSenseCustomerSupport Hi, I still have two questions.
1. The [link](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md) is not prepared for python wrapper. 
2. The [link](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python) has no instruction on installing python wrapper on MacOS.

So the question is how to build the python wrapper from source in mac os environment?

syh4661, 2024-04-26:
Thank you for replying.
Anyway, I changed my OS.

Thank you for your full consultant.