realsenseai/librealsense #3261: Realsense SDK2.0 build in anaconda in linux

sohn21c, 2019-02-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi,

I'm in need to install the SDK2.0 on my anaconda environment. I searched the list and saw the posting related to the windows but wanted to see how I should go about doing it in linux. Please see the information below.

Camera: D435
OS: Ubuntu 18.04
Kernel version: 4.18.0-15-generic
Platform: PC
SDK version: 2.0 (2.18.1)
Language: Python

I had no issue installing it with dpkg on my machine but I am not familiar with the process of building it from the source so I can engage the SDK to anaconda.

MartyG-RealSense, 2019-02-14:
I hope the link below will be helpful to you.

https://sebastianwallkoetter.wordpress.com/2018/02/18/realsense-sdk-in-anaconda-3/

sohn21c, 2019-02-14:
It came up on my previous search as well but he's one window using Cmake gui which isn't available on the linux?

MartyG-RealSense, 2019-02-14:
Googling for 'cmake GUI Linux' provides a number of leads.  Example:

https://askubuntu.com/questions/1020168/how-to-build-cmake-gui-on-linux-from-source

FirefoxMetzger, 2019-02-15:
> It came up on my previous search as well but he's one window using Cmake gui which isn't available on the linux?

I was actually doing it on Linux (Ubuntu 16.04). It has the cmake-gui as well, though it comes as a separate package. On 16.04 you can install it via `sudo apt install cmake-gui`. 

I haven't upgraded to 18.04 yet (manly due to laziness and fear of breaking existing configurations on my machine). I don't know if the cmake-gui package exists there yet, but the cmake-qt-gui should (which is the same thing, iirc). So, you could try `sudo apt install cmake-qt-gui` to install it and then use `cmake-gui` to start it :)

sohn21c, 2019-02-16:
Thanks for the information. As soon as I saw the gui, i assume it was done in windows.
But still, as I'm new to all this the explanation is a bit sparse. Do you think you can elaborate? I installed the cmake-gui and can now launch. I have created environments in the anaconda before but how should I go about configuring all this? I don't even see any dropdowns as I launch the cmake-gui. Should I cd to certain directory? Should I have downloaded something?

Thank you all for your help.

MartyG-RealSense, 2019-02-16:
Is this list of commands for cmake-gui any help, please?

https://cmake.org/cmake/help/v3.14/manual/cmake-gui.1.html

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

@sohn21c Can you launch cmake-gui as MartyG-RealSense's guide? If the installation of cmake-gui has no problem,you should get the GUI when type the command "cmake-gui".

closed 2019-02-21 by sohn21c