realsenseai/librealsense #3262: question of multi-cam with D415s

sbnn123456, 2019-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { {D400}  | 
| Firmware Version                   | (not the newest one) | 
| Operating System & Version |   {Win (10)}  | 
| Kernel Version (Linux Only)    |  (not using)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {}                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi dear sir

I have two D415 cams needed to be synced. I am wondering that belows:

in paper of using the intel realsence depths camera D4xx in multi-camera configurations,
the paper writes "rs2_Set_Option with argument “rs2_option_inter_cam_sync_mode” "

is it can be used in the latest firmware and SDK now?

any response will be appreciated

MartyG-RealSense, 2019-02-15:
Hi sbnn123456,

Multi-camera sync of the D415 has been available since June 2018.

sbnn123456, 2019-02-15:
@MartyG-RealSense thanks a lots.

i was seen two way to sync multi-cam ,one is HW Sync.
you means it also can be used by HW sync now?

maybe same question asked again. thank you very much.

MartyG-RealSense, 2019-02-15:
Yes, there is software sync and hardware sync.  HW sync is available now.

sbnn123456, 2019-02-15:
@MartyG-RealSense  thank you very much.
we really appreciate you help.

closed 2019-02-15 by sbnn123456