realsenseai/librealsense #3263: D435i device register failed when connected

yerzhik, 2019-02-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.1.0 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |   4.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.1                          | 
| Language                            |  C                          | 
| Segment			|  Robot                          | 

### Issue Description

> When attached to PC, the dmesg shows:
> uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> uvcvideo: Unable to create debugfs 2-21 directory.
> uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> uvcvideo 2-6:1.0: Entity type for entity Processing 2 was not initialized!
> uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> uvcvideo 2-6:1.0: Entity type for entity Camera 1 was not initialized!
> input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input230
> uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> uvcvideo: Unable to create debugfs 2-21 directory.
> uvcvideo 2-6:1.3: Entity type for entity Processing 7 was not initialized!
> uvcvideo 2-6:1.3: Entity type for entity Extension 8 was not initialized!
> uvcvideo 2-6:1.3: Entity type for entity Camera 6 was not initialized!
> ------------[ cut here ]------------
> sysfs: cannot create duplicate filename '/dev/char/235:0'
> ... etc ...
> hid_sensor_gyro_3d HID-SENSOR-200076.2.auto: device register failed
> hid_sensor_gyro_3d: probe of HID-SENSOR-200076.2.auto failed with error -17

realsense-viewer won't find device, but connecting D435 (without i) works fine. 
[Full dmesg log](https://pastebin.com/JbLwgETw)

All steps done by firmware update and installation manual.

Could you help with this problem?

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi yerzhik,

Can you please try to applcy udev rules again and then check it?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`
`sudo udevadm control --reload-rules && udevadm trigger`

Thanks!

yerzhik, 2019-03-01:
I did that before but I missed running the patch. Thank you!

closed 2019-03-01 by yerzhik

Ahmad-Drak, 2020-07-09:
I have a relatively similar error but on ubuntu 18.04  5.3.0-53-generic. After connecting the cam dmesg output:

> [  682.342781] usb 4-1.3: new SuperSpeed Gen 1 USB device number 4 using xhci_hcd
> [  682.367545] usb 4-1.3: New USB device found, idVendor=8086, idProduct=0b3a, bcdDevice=50.c5
> [  682.367552] usb 4-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  682.367556] usb 4-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435i
> [  682.367560] usb 4-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
> [  682.367564] usb 4-1.3: SerialNumber: 020323050614
> [  682.382042] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> [  682.388597] uvcvideo 4-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> [  682.388598] uvcvideo 4-1.3:1.0: Entity type for entity Processing 2 was not initialized!
> [  682.388599] uvcvideo 4-1.3:1.0: Entity type for entity Camera 1 was not initialized!
> [  682.388656] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:1c.1/0000:07:00.0/usb4/4-1/4-1.3/4-1.3:1.0/input/input36
> [  682.390550] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> [  682.400423] uvcvideo 4-1.3:1.3: Entity type for entity Processing 7 was not initialized!
> [  682.400425] uvcvideo 4-1.3:1.3: Entity type for entity Extension 8 was not initialized!
> [  682.400426] uvcvideo 4-1.3:1.3: Entity type for entity Camera 6 was not initialized!
> [  682.562478] hid-sensor-hub 0003:8086:0B3A.000D: No report with id 0xffffffff found

I tired the suggestion below but it didnt help.

> Hi yerzhik,
>
> Can you please try to applcy udev rules again and then check it?
>
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites
>
> `sudo cp [...]

Any idea what causes this?

Ahmad-Drak, 2020-07-09:
If i run realsense-viewer after connecting the cam i get this in dmesg (that keeps repeating):

> [ 1152.170764] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ 1152.291817] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ 1152.433812] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ 1152.488874] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ 1152.718472] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
> [ 1163.438701] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

In real-sense viewer i try to do a firmware update, the cam disconnects at this output:

> Started Firmware Update process
> Backing-up camera flash memory
> Backup completed and saved as '/home/ahmad/.020323050614.20200709_191406.bin'
> Requesting to switch to recovery mode
> 
> ERROR: Recovery device did not connect in time!

when that happens dmesg shows this:

> 
> [ 1223.762220] xhci_hcd 0000:07:00.0: Abort failed to stop command ring: -110
> [ 1223.778231] xhci_hcd 0000:07:00.0: Host halt failed, -110
> [ 1223.778232] xhci_hcd 0000:07:00.0: xHCI host controller not responding, assume dead
> [ 1223.778239] xhci_hcd 0000:07:00.0: xHCI host not responding to stop endpoint command.
> [ 1223.794249] xhci_hcd 0000:07:00.0: Host halt failed, -110
> [ 1223.794253] xhci_hcd 0000:07:00.0: xHCI host not responding to stop endpoint command.
> [ 1223.810264] xhci_hcd 0000:07:00.0: Host halt failed, -110
> [ 1223.810280] xhci_hcd 0000:07:00.0: HC died; cleaning up
> [ 1223.810298] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -110 (exp. 1024).
> [ 1223.810563] uvcvideo: Failed to query (GET_CUR) UVC control 7 on unit 3: -19 (exp. 1).
> [ 1223.811798] usb 3-1: USB disconnect, device number 2
> [ 1223.812073] usb 4-1: USB disconnect, device number 2
> [ 1223.812075] usb 4-1.3: USB disconnect, device number 4
>

Ahmad-Drak, 2020-07-09:
If i run realsense-viewer after connecting the cam i get this in dmesg (that keeps repeating):

> [ 1152.170764] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ 1152.291817] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
> [ [...]

In real-sense viewer i try to do a firmware update, the cam disconnects at this output:

> Started Firmware Update process
> Backing-up camera flash memory
> Backup completed and saved as '/home/ahmad/.020323050614.20200709_191406.bin'
> Requesting to switch to recovery mode
> 
> ERROR: Recovery device did not connect in time!

when that happens dmesg shows this:

> [ 1223.762220] xhci_hcd 0000:07:00.0: Abort failed to stop command ring: -110
> [ 1223.778231] xhci_hcd 0000:07:00.0: Host halt failed, -110
> [ 1223.778232] xhci_hcd 0000:07:00.0: xHCI host controller [...]

Ahmad-Drak, 2020-07-11:
@RealSenseCustomerSupport can you help with this?