realsenseai/librealsense #3264: set_frame_by_index

RZz96, 2019-02-15:
how to read _frame_by_index using python. im reading frame from bag file.

MartyG-RealSense, 2019-02-15:
For the SR300, D415 and D435 cameras and the current RealSense SDK 2.0, the link below is the best available discussion about accessing frames by index with Python.

https://github.com/IntelRealSense/librealsense/issues/3121

**set_frame_by_index** seems to be an instruction from the RealSense SDK For Linux.  

https://software.intel.com/sites/products/realsense/sdk/classrs_1_1playback_1_1device.html

This is an older SDK written specially for the RealSense ZR300 camera model, which is not compatible with the SR300, D415 and D435.  Are you using a ZR300 please?

RZz96, 2019-02-15:
Thanks for reply 
im working on windows and using D435.
need to pause the frame go to pre frame and next frame 
any help?

MartyG-RealSense, 2019-02-15:
@dorodnic the RealSense SDK Manager says in that linked discussion that "jumping back and forward is more complicated, and there are possibly additional bugs hidden there.  Simplest way is to reset to start, and skip until the desired frame".

Going to the previous and next frame would seem to qualify as 'jumping back and forward'.  Hopefully the Intel guys such as Dorodnic can provide some further suggestions for how you might implement a skip back-forward function.

RZz96, 2019-02-15:
thank you for the reply .
would you please tell me for now , by pressing key how can i  pause reading the frame . what is the function for pause. they said playback.pause() but : 
device= pipeline.get_active_profile().get_device()
Playback=device.playback()
get an error :
'pyrealsense2.pyrealsense2.device' object has no attribute 'playback'

MartyG-RealSense, 2019-02-15:
It's a slightly complicated subject.  Intel support team member Eliza has suggested to someone else seeking to implement a pause playback function that they visit the example Python program **read_bag_example.py**  linked to below, especially focusing on the While loop at line 52.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py?language=en_US#L52

Eliza says "The "while True" loop beginning on line 52, loops through each frame 1 by 1.  You can do whatever treatment you want on each frame within this loop".

RZz96, 2019-02-15:
im using the same code but i need to use playback to pause and resume but i cannot have the playback .

i dont know what is wrong

MartyG-RealSense, 2019-02-15:
RealSense programming is not one of my specialist areas unfortunately, so one of the Intel guys will be better suited to providing help with this question.  I do apologize.

lramati, 2019-02-16:
Use device.as_playback() not device.playback()

dorodnic, 2019-02-20:
Closing as duplicate.
Controlling playback at the frame level is marked as a desirable enhancement to the SDK, but at the moment it is beyond the scope of the API.

closed 2019-02-20 by dorodnic