realsenseai/librealsense #3266: Viewer exits when attempting to record

BW25, 2019-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Dell Laptop  |
| SDK Version                            | 2.17.1                          | 
| Language                            |                           | 
| Segment			|                            | 

### Issue Description
When using the Viewer, attempting to record any video causes the viewer to exit.
This occurs when the Record button is clicked.
This occurs regardless of which camera is selected (D435i or laptop's integrated webcam).
It is still capable of taking snapshots.

dorodnic, 2019-02-15:
Hi @BW25 
What happens when you run the application as an administrator? (I suspect there is some permission problem with the default recording folder)

BW25, 2019-02-15:
> Hi @BW25
> What happens when you run the application as an administrator? (I suspect there is some permission problem with the default recording folder)

The same result, unfortunately.
I have also tried it with SDK 2.17.0, with the same result.

BW25, 2019-02-15:
Issue resolved, although I am unclear as to the cause.
I switched back and forth between advanced mode and found afterwards, the issue was no longer present.
Thank you.

closed 2019-02-15 by BW25