realsenseai/librealsense #3267: D435i not recognized in Unity when

andreisperid, 2019-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0                       | 
| Language                            |  unity                           | 
| Segment			|  VR                 | 

### Issue Description
Hello, Ive downloaded the Unity wrapper (everything indicates that it already contains both required DLLs from SDK)

If I run the "StartHere" scene in Unity with the D435i plugged in USB 2.1 input, it runs alright but hides some options (I guess that it is because full capabilities has not been unlocked because it isnt USB 3.0)
[image]
[image]

If I run it plugged on USB 3.2 port, it is not recognized.
[image]

If I unplug and plug again with the scene running, I receive this error on console:
[image]

Thank you

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi andreisperid,

I tried to compile from source off RS 2.17.0 (as I have this release source for some other work right now).
There is no issue to use the unity samples with D435i plugged in USB3 port (verified by adjusting to stream depth+IR 1280x720p 30fps).
And also no issue with D435i plugged in USB2 port (verified by limited to stream capability vs. USB3 port).

Not sure what you could be missing. Can you please try again?

Thanks!

andreisperid, 2019-02-25:
Hi!

I actually noticed I was using 2.18.1 instead of 2.17.0 and forgot to come back to update (fixing it right now).

Anyways, for 2.18.1 it isnt working

Thank you!

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi andreisperid,

Thanks for confirming that RS 2.17.0 works for you.

I checked out RS 2.18.1 source code and repeated the tests with D435i. Again, there has no issue on recognizing D435i off USB3 port & USB2 port.
I am using Unity 2018.1.13f1 64-bit personal release.

Thanks!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi andreisperid,

Wonder if there is any update? If nothing else is needed, this one will be closed soon.

Thanks!

andreisperid, 2019-03-07:
Hi! Sorry for not answering. Ive updated to the 2.19, both SDK and unity assets and its running smoothly! 

Im actually using a 435 and a 435i together inside of Unity and no issues.

Thank you for you attention!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi andreisperid,

Thanks for the updates.
Glad to hear that it works for you now with RS 2.19.0 release.
I am closing this one.

Thanks!

closed 2019-03-14 by RealSenseCustomerSupport