realsenseai/librealsense #3269: D435 has no image output(dark screen, 0 fps) when using chrome browser in ubuntu 16.04

ATMRORO, 2019-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00| 
| Operating System & Version | Linux (Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    | 4.15.0-45-generic #48~16.04.1-Ubuntu SMP)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.18.1  | 
| Language                            |  node.js/python   | 
| Segment			|  Robot   | 

### Issue Description
Does anybody get images successfully from D435 that using chrome browser(version 71.0.3578.98) in ubuntu 16.04? D435 has no image output(dark screen, 0 fps) when using chrome browser in ubuntu 16.04.
I try it on windows 10. it is workable, but not in ubuntu. Chrome browser get three camera, but no one can get images. Just dark screen. 
"Intel(R) RealSense(TM) Depth Camera 435 with RGB Module Depth (8086:0b07)",
"Intel(R) RealSense(TM) Depth Camera 435 with RGB Module Depth (8086:0b07)",
"Intel(R) RealSense(TM) Depth Camera 435 with RGB Module RGB (8086:0b07)"

realsense-viewer, rs-capture , python example codes, these are all workable on my ubuntu system....

below is my device information:
robert@robert-house:~/librealsense/build/examples/capture$ dkms statusbbswitch, 0.8, 4.15.0-29-generic, x86_64: installed
bbswitch, 0.8, 4.15.0-43-generic, x86_64: installed
bbswitch, 0.8, 4.15.0-45-generic, x86_64: installed
bbswitch, 0.8, 4.4.0-142-generic, x86_64: installed
librealsense2-dkms, 1.3.4, 4.15.0-45-generic, x86_64: installed
nvidia-410, 410.79, 4.15.0-29-generic, x86_64: installed
nvidia-410, 410.79, 4.15.0-43-generic, x86_64: installed
nvidia-410, 410.79, 4.15.0-45-generic, x86_64: installed
nvidia-410, 410.79, 4.4.0-142-generic, x86_64: installed
robert@robert-house:~/librealsense/build/examples/capture$ modinfo uvcvideofilename:       /lib/modules/4.15.0-45-generic/updates/dkms/uvcvideo.ko
version:        1.1.2.realsense-1.3.4
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     C4FB5E9B212E9C39ED6CD7C
alias:          usb:v*p*d*dc*dsc*dp*ic0Eisc01ip01in*
alias:          usb:v*p*d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B3Ad*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B0Dd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B0Cd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AA5d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B07d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B03d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B01d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0B00d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AFFd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AFEd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AF6d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AF2d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AD5d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AD4d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AD3d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AD2d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v8086p0AD1d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v2833p0211d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v2833p0201d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1C4Fp3000d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1B3Bp2951d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d00*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d01[0-1]*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d012[0-6]dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v199Ep8102d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v18ECp3290d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18ECp3288d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18ECp3188d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18CDpCAFEd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1871p0516d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v1871p0306d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v17EFp480Bd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v17DCp0202d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp8A34d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp8A33d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp8A31d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp8A12d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp5931d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v174Fp5212d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v152Dp0310d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v13D3p5103d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0E8Dp0004d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0BD3p0555d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0AC8p3420d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0AC8p3410d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0AC8p332Dd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v06F8p300Cd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05E3p0505d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05C8p0403d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05ACp8600d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05ACp8501d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05A9p7670d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05A9p264Ad*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05A9p2643d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05A9p2641d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v05A9p2640d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v058Fp3820d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v04F2pB071d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v046Dp082Dd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v046Dp08C7d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v046Dp08C6d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v046Dp08C5d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v046Dp08C3d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v046Dp08C2d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v046Dp08C1d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v045Ep0723d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v045Ep0721d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v045Ep00F8d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0458p706Ed*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v0416pA91Ad*dc*dsc*dp*ic0Eisc01ip00in*
depends:        videodev,videobuf2-core,videobuf2-v4l2,videobuf2-vmalloc,media
retpoline:      Y
name:           uvcvideo
vermagic:       4.15.0-45-generic SMP mod_unload 
parm:           clock:Video buffers timestamp clock
parm:           hwtimestamps:Use hardware timestamps (uint)
parm:           nodrop:Don't drop incomplete frames (uint)
parm:           quirks:Forced device quirks (uint)
parm:           trace:Trace level bitmask (uint)
parm:           timeout:Streaming control requests timeout (uint)

MartyG-RealSense, 2019-02-16:
Advice about using the 400 Series with browsers is provided in the discussion linked to below.

https://github.com/IntelRealSense/librealsense/issues/1764

closed 2019-03-07 by RealSenseCustomerSupport

PhillipVoyle, 2019-07-21:
@ATMRORO did you find a fix for this? I have a D435, and am experiencing the same problem. The link above talks about using the depth info, which isn't useful to me for my immediate purposes

ATMRORO, 2019-07-22:
No... I given up to using it. I use an usb webcam for chrome browser now.....

PhillipVoyle, 2019-07-22:
@ATMRORO Ok, thanks for your response. I might end up with that